Ultimaker/Cura #7563: Top/Bottom Layer count miscalculated

Sophist-UK, 2020-04-22:
**Application version**
4.6.0

**Platform**
Windows 10

**Printer**
Dagoma DiscoEasy200

**Reproduction steps**
1. Changed Top/Bottom layer count and then clicked the Fx button to recalculate defaults.

**Actual results**
For top / bottom layer thickness of .4mm (with Fine profile of 0.1mm layer height):

Top Layers: 0
Bottom Layers: 999999

**Expected results**
Top Layers: 4
Bottom Layers: 4

Sophist-UK, 2020-04-22:
No - 10% Gira.

Sophist-UK, 2020-04-22:
It didn't happen in 4.5.0.

rburema, 2020-04-23:
I could not get this to reproduce (with anything else than 100% infill set that is) . Could you send a project file, so we can check what's going on? (Save, don't export, rename in order for github to accept it.)

closed 2020-06-19 by no-response[bot]