Ultimaker/Cura #7564: Initial Layer Height doesn't apply to brim / skirt

Sophist-UK, 2020-04-22:
**Application version**
4.6.0

**Platform**
Windows 10

**Printer**
Dagoma DiscoEasy200

**Reproduction steps**
1. Start to print something with a brim using Draft / Fast profile. Stop it once the brim has printed.
2. Print it again using Fine settings.

**Screenshot(s)**
Sorry - not possible to capture as brim is so thin as it is invisible.

**Actual results**
Brim on fine settings is practically non-existent.

**Expected results**
Brim on Fine profile  should be same as on Draft / Fast profile because the initial layer height is the same.

I suspect (but it's just a guess) that the Initial Layer Height is NOT being applied to Brim (and I suspect equally to Skirt).

rburema, 2020-04-23:
Is it just the brim/skirt, or actually the first few layers? When I slice with that model, then load back the gcode (with the 'layer thickness' colour scheme), I can see 3 strange layers, before the model starts.

It only looks a _little_ bit better when I do that in Draft.

Since this doesn't seem to happen on other printers...
My best guess is that those profiles aren't doing what they should. It seems they have been updated in the beginning of this year by a contributor. @0r31 Can you tell if this is the profiles, or am I mistaken and is the bug somewhere else?

0r31, 2020-04-23:
Hi,

Could you please send a screenshot?
Actually, i tried to introduce bicolor profile but was not aware of some stuff regarding extruder start gcode usage in recent CuraEngine version.

I advise you to comment (or remove) all the extruder start gcode sequence in your printer profile:
[image]

Regards,
Orel

Sophist-UK, 2020-04-23:
I have not previously had the chance to update this issue today, but I noticed that the filament had been retracted a HUGE amount - possibly 5cm or more - and that was the cause of the brim not printing properly.

I had also noticed that the standard DiscoEasy200 profile was set with 2x extruders, and had to change it to one - whereas I believe that one extruder should be the default with 2 as an option.

But the above code might well explain what is happening. If I had a dual head I would expect this amount of retraction when changing filament.

I have no idea what needs to happen with other printers, but it seems to me that for Extruder 1 you need Start and End Gcode when it is the only extruder, and different start and end code when it is the first of two or more extruders.

0r31, 2020-04-23:
Hi @Sophist-UK ,

Agree.
This huge amount of retract is set in the extruder start gcode (no the global start gcode) and is intended to be used for the bicolor option to remove the filament from the nozzle and just let enough place to use the other one.
As far as i understand these extruder change gcodes sequences should occur **only** when the extruder is changed. I can tell why but it seems it is always added after the global start gcode (cf. https://github.com/Ultimaker/CuraEngine/blob/071a5e60c3b19959cb718bb4b4991cff44f6159e/src/FffGcodeWriter.cpp#L585). As i am pretty sure there is a good reason to do so, the goal is to adapt the profile regarding this stuff. I will do my best to update the profile (and as you said, if no "easy solution", the quick and dirty solution is to have two different profiles for single/dual extrusion. It doesn't solve the dual extrusion but it solves the single one ;)

Regards,
Orel

Sophist-UK, 2020-04-23:
If you have a dual extruder, then you will either:

A, always need to retract the filament at the end of the print in case the next print only uses the other extruder; or
B. always leave the printer with filament from Extruder 1 in the HotEnd and Extruder 2 retracted.

See also #7562 - it also occurs to me that if you cancel a (USB) print job, then not only should you send the Machine End Gcode, you should also leave the extruders in the same state they would be in if the print had run to completion.

Sophist-UK, 2020-04-23:
@0r31 - Do we need two separate definitions for DE200 - one with one extruder and one with two?

Sophist-UK, 2020-04-24:
On further reflection, if you only have a single extruder then there is little point for Extruder start/end Gcode. So perhaps Cura should only send the start/end code if there is more than one extruder and if there is only one extruder the fields in Machine Settings should be disabled.

0r31, 2020-04-24:
> If you have a dual extruder, then you will either:
> 
> A, always need to retract the filament at the end of the print in case the next print only uses the other extruder; or
> B. always leave the printer with filament from Extruder 1 in the HotEnd and Extruder 2 retracted.
> 

The A solution is the chosen one (already done in the soft provided by Dagoma -a fork from the old Legacy Cura-).

I am aware there is still some work to make profiles fully efficient.

0r31, 2020-04-24:
> @0r31 - Do we need two separate definitions for DE200 - one with one extruder and one with two?

I can't tell at this point. It needs to be investigated to find the right way to manage the current Cura behavior. Maybe there is no smart solution so the final one would be to have two separate profile. I don't have the answer yet.

Sophist-UK, 2020-04-24:
I think the easiest is to have separate profiles for printers with 1 and more than 1 extruder.

Any other solution will be confusing for users.

Ghostkeeper, 2020-04-28:
> As far as i understand these extruder change gcode sequences should occur only when the extruder is effectively changed. I can't tell why but it seems it is always added after the global start gcode (cf. https://github.com/Ultimaker/CuraEngine/blob/071a5e60c3b19959cb718bb4b4991cff44f6159e/src/FffGcodeWriter.cpp#L585).

We consider this a bug: https://github.com/Ultimaker/Cura/issues/6160

Fixing that sort of thing is a bit involved though since it could break printers that were added after the bug was introduced and now rely on this bug.

0r31, 2020-04-28:
Hi @Ghostkeeper ,

Thanks for the update. Is there a new version incoming? If so, i can make changes on Dagoma printers profile (disabling the dual extruder feature for now in waiting to have a proper solution.

Ghostkeeper, 2020-04-28:
There is a 4.6.1 version planned, but no promises on when or even whether it will be released. If it's released we won't accept any PRs into that version though since that would need re-testing and we need this to be as stable as possible.

After that, there will be about 9001 new versions of Cura. The next one is planned in about 2 months and we can accept your profile changes for that one.

Sophist-UK, 2020-04-28:
As I said previously, the obvious thing is to have separate profiles for DE200 with a single extruder and with multiple extruders.

rburema, 2020-04-28:
It would certainly be the easiest and cleanest way, yes.

0r31, 2020-04-28:
Hi,

It is certainly the easiest way but not the cleanest one imho.
Anyway, i will submit a PR in this way asap (separate single/dual extrusion printer profile).

Regards,

Bigsmooth68, 2020-05-04:
Hi,
I have 4.6.1 and it still has the issue. Was it postponed to a further release ?
thanks for the work !
Regards,
Olivier

0r31, 2020-05-05:
Hi @Bigsmooth68 ,

Please check the last message from @Ghostkeeper : To sum up, no new PR merged for the 4.6.1 release.

Regards,

kirax999, 2020-05-06:
I also updated to 4.6.1 and I actually have a big problem with extruding too much is not enough I am trying to remove the start and end of GCODE. I try again with cura 4.4, it works very well, so I made a backup of the configuration and I try to download it again in 4.6.1 and not x) cura tells me that the profile is corrupted, so i will use 4.4

And I'm waiting for the updates from Cura

Ghostkeeper, 2020-05-07:
We do accept PRs for 4.7 though.

galax-ajh, 2020-06-07:
Hi.
I have encounter this issue. No need to have 2 profiles. Start and End gcode are inverted. The good gcode are :
Extrudeur 1 start gcode
`;Start T0
G92 E0
G1 E{retraction_amount} F3000
G92 E0
G1 E60 F3000
G92 E0
G1 E-{retraction_amount} F5000`
Extruder 1 end gcode:
`G92 E0
G1 E-{retraction_amount} F10000
G92 E0G1 E1.5 F3000
G1 E-60 F10000
G92 E0
;end T0`
For extruder 2, apply same gcode (juste think to change T0 -> T1 in commented lines.
the only issue in that case, is that you are extruding 60 mm wich are not needed.

0r31, 2020-06-10:
Hi @galax-ajh ,

Nice catch ! I was not aware of that and just naively set preswitch (resp. postswitch) sequence from 15.04 as extruder start (resp. end) gcode.

I just checked settings mapping between 15.04 and 2.0 (available [here](https://github.com/Ultimaker/CuraEngine/blob/08fe61a7d28ddd08603a9d9e3e625d31e8ed8071/meta/15.04_to_2.0.txt#L42) and you are totally right.

My bad...

I won't gather mono/dual extruders profiles at this point but i am about to switch extruder gcodes sequences.

Many thanks,
Orel

0r31, 2020-06-10:
Hi again @galax-ajh ,

I just pushed changes in the related PR : #7793.
Don't hesitate to test/come back to me if you see any issue.

Regards,
Orel