Ultimaker/Cura #7565: Monitor tab Send Gcode - display responses

Sophist-UK, 2020-04-22:
**Is your feature request related to a problem? Please describe.**
If you use the send Gcode box on the monitor tab, any responses are not displayed.

**Describe the solution you'd like**
Display any responses sent by the printer.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
USB (and network?) printers.

Ghostkeeper, 2020-04-28:
This would be a nice new feature for the monitor, but we'll have to defer this. This sort of feature is very advanced, touching in the firmware and debugging routines of your printer. I know that there are such commands out there to get the current position of the print head and such, but I don't know any commands by heart myself even aside from the temperature probe commands `M105` which is already displayed in that monitor stage.

I think for now you'll have to resort to Pronterface or somesuch.

Sophist-UK, 2020-04-28:
Note: mention in issue 7611 was a typo which has since been corrected.

Sophist-UK, 2020-04-28:
It is clear that the printer sends information back to Cura so that it can display e.g. the print head temperature. So this shouldn't be too hard to code.

And to clarify the requirement:

I am asking for responses to be displayed ONLY when the user has sent manual Gcode to the printer using the input box provided - not displaying the responses received when it is sending a sliced print to the printer.

(P.S. It would be nice to have the manual Gcode box having a dropdown list of Gcodes, but really that would be overkill. If you are sending manual Gcode, then it can be assumed that you know what you are doing!! I don't know Gcode off by heart either but the Marlin and RepRap sites provide reference pages of these.)

fieldOfView, 2020-04-28:
> So this shouldn't be too hard to code.

It is not hard to code, but it is irrelevant to all Ultimaker printers except the Ultimaker Original. And that is such a small portion of the Ultimaker revenue that it is not worth spending the time for Ultimaker developers.

> I am asking for responses to be displayed ONLY when the user has sent manual Gcode to the printer using the input box provided

That actually makes it harder though; your printer does not say what it is responding to when it sends information back to the computer.

Sophist-UK, 2020-04-28:
P.S. Quote from the [RepRap Gcode page](https://reprap.org/wiki/G-code):
> Hosts normally monitor printer temperatures by sending M105 every couple of seconds. This not only adds serial traffic but it will fail whenever the command queue is full. M155 addresses these problems by telling the firmware to automatically report temperatures at regular intervals. This behavior is disabled by default for best compatibility with existing hosts. If the firmware supports M155 the output of M115 will report the AUTOREPORT_TEMP capability:
>```
>Cap:AUTOREPORT_TEMP:1
>```

fieldOfView, 2020-04-28:
AUTOREPORT_TEMP is one of the things making it harder to know what response goes with what command, without knowing what the response should look like.

Sophist-UK, 2020-04-28:
Yes - I accept that this is not going to be a priority for Ultimaker - so it will need to be a community PR.

We would need to use a timer of xxxxms and simply show what is sent back in that period that isn't an AUTOREPORT_TEMP formatted responme.

Ghostkeeper, 2020-04-28:
> It is clear that the printer sends information back to Cura so that it can display e.g. the print head temperature. So this shouldn't be too hard to code.

For that feature Cura explicitly sends M105 commands in between and parses the responses, yes. Cura can read the responses to be sure, and waits on these responses continuously to prevent sending data too quickly to the printer when printing long lines. The implementation you're really asking for is to display the results of these commands in the interface. I'm not saying it's all that much work. It's just work that we can't do now.

Making it only display the responses of manual g-code commands is a bit more difficult though. You'd need to count commands.

closed 2023-07-13 by github-actions[bot]