Ultimaker/Cura #7569: Track pad on laptop activates zoom, despite feature turned off in preferences

Pete-Snell, 2020-04-22:
Hi

   Win 7 CF-52 Toughbook Cura 4.6. 

  Track pad (mouse) zoom feature active although it is shut off in preferences. This makes moving the cursor aggravating, as the image keeps zooming in and out as you are working.

Pete

mahtDFR, 2020-04-23:
That issue template you deleted to post this? Please fill it out. I have no information of what version of Cura you're using.

mahtDFR, 2020-04-23:
"Zoom towards mouse direction" right? Just tried in Win10 and it still zooms after disabling it but just not towards where the cursor is positioned. I'm having to make some assumptions here to try and help you so next time fill out that issue template please.

rburema, 2020-04-23:
I think we can consider this a feature request then: Turn of zoom _entirely_ as a preference (for trackpad users).

mahtDFR, 2020-04-24:
We're probably not going to implement a feature that disables zoom entirely.
You can hold the spacebar and use one finger on the trackpad to zoom. If you don't want to zoom, don't use two fingers.

closed 2020-04-28 by Ghostkeeper