Ultimaker/Cura #7570: "Spot" brims

KenLucke, 2020-04-22:
First off, AGAIN, please add the CR-10v2 to your pre-installed printer profiles.  It's a HUGELY popular printer, and it's parameters are not quite the same as the 10 or 10s you already have.  I've seen SEVERAL new versions of Cura come out and it has YET to add this printer, despite it having been on the market for a long time now.

As to the feature request, it would be something similar (or a setting within) the support blocker/modify settings:  A "spot" brim that could be added at critical spots where prints are likely to lift off the bed, such as the end of long, thin features, or corners prone to warping.  This would make it possible to just put a brim in those spots rather than having to do it for the full build, wasting time, filament, and effort where it's not needed.

mahtDFR, 2020-04-23:
We don't make machine definitions for for third party printers, they come in via contributors. We can't maintain these profiles as we don't have the time and means to test them. If you'd like to see this printer in Cura, please either make a pull request with a definition for this, or encourage someone else to do so.

As for your FR its something that's been considered already, I will discuss and see what we think.

BagelOrb, 2020-07-08:
This is a duplicate of the mouse ears feature request https://github.com/Ultimaker/Cura/issues/1349

closed 2020-07-08 by BagelOrb