Ultimaker/Cura #7572: [4.6.0] Ironing appearing in/on walls

Nemernemer, 2020-04-22:
**Application version**
Ultimaker Cura 4.6.0

**Platform**
OS:  Microsoft Windows 10 Pro (10.0.18363)
CPU:  i7-9700K @ 5.10GHz
RAM:  4x8GB DDR4 Vengeance Pro 3200MHz
GPU:  RTX 2070S
MB:  ROG STRIX Z390-F GAMING

**Printer**
Creality - Ender 5

**Reproduction steps**
1. Slice model with ironing enabled (all surfaces) in 4.5.0
2. Slice model with ironing enabled (all surfaces) in 4.6.0 (Same profile as included with this bug report)

**Screenshot(s)**
4.5.0:
[image]

4.6.0:
[image]

**Actual results**
In 4.6.0 you can see ironing lines in the hole that weren't there in the 4.5.0 slice with the same settings. These disappear if Ironing is turned off / set to only top layer.

**Expected results**
Ironing not taking place in vertical skin.

**Project file**
[Project file.zip](https://github.com/Ultimaker/Cura/files/4519358/Project.file.zip)

**Log file**
Apologies, but I don't know which parts of this is relevant or not, so I'll add the whole thing:
[cura.log](https://github.com/Ultimaker/Cura/files/4519363/cura.log)

rburema, 2020-04-23:
Please take a look at the 'Ironing Inset' setting.

Nemernemer, 2020-04-23:
Problem goes away when ironing inset reaches 0.46 mm. (This is with a line width of 0.44 mm.)
Lowering the layer height also reduses the amount of artifacts around the print. (Going from 0.2 to  0.12 and you have to start looking for them. All artifacts in holes are gone.)

But I consider this a temporary fix, as adjusting these values were not necessary in Cura 4.5.0.

Ghostkeeper, 2020-04-28:
Interesting! This reproduces in 4.6.0 but not in 4.6.1 or 4.7.0, suggesting that it was also caused by a fix that we made to the simplify algorithm recently. My theory is that reducing the resolution of the model causes there to be a tiny corner there which gets amplified somehow through the various offsets applied.

That fix is reverted for the (probably) upcoming 4.6.1 and properly re-fixed in master. In neither of these the bug occurs any more. Thanks for reporting it though!

closed 2020-04-28 by Ghostkeeper