Ultimaker/Cura #7574: Z Offset Plugin shows installed  in Cura 4.5 & 4.6 but not available in Build Plate Adhesion

H-DieterK, 2020-04-23:
**Application version**
Both Cura 4.5 and 4.6 Tried reinstalling fresh downloads of both versions to work around issue.

**Platform**
Window 10 latest update

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Installed per Marketplace Plugin process.  Showed as installed when Cura reloaded, but did not show up as expected in either version of Cura  in Build Plate Adhesion Setting regardless of subselections.
2. Attempted to delete from Plugins several times after exiting and relaunching Cura's.  Never deleted the plugin and showed as still installed.  Could not reinstall a fresh Z offset plugin!

**Screenshot(s)**
None.  Shows up in installed Plugin list.  Does not appear in Adhesion settings.

**Actual results**
As above note.

**Expected results**
Expected that a fresh plugin would overwrite the corrupted file and appear in Settings.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-04-23:
I can't help you without a log file.

H-DieterK, 2020-04-24:
Hello Aldo
Thanks for getting back.
How do I find the log file?
Hans

fieldOfView, 2020-04-24:
As the issue template says, see https://github.com/Ultimaker/Cura#logging-issues

Go to Help -> Show configuration folder. Locate the file named "cura.log".

H-DieterK, 2020-04-24:
Hi Aldo,
Yes I did find the Cura.log file.
I found the missing link in the expanded Setting Visibility Preferences, well down the list after I had changed from Expert selection to All.
It;s somewhat counter intuitive to toggle between All and Expert modes when the respective ‘Z Offset’ options have been selected in All mode not to have this option also selected in Expert.
Even when selecting Check all in the top of Setting Visibility panel, it does not behave as expected as there remain numerous unselected detail options that appear.
Nowhere on imbedded instructions for Cura and Plugin behaviours does it mention how to find these missing detail features, even when searching with the ‘Filter’ box.
The Marketplace Plugin activations mostly appear automatically when relaunching Cura 4.5 or 4.6 either in Windows or Mac platforms.
I was expecting to find Z Offset options in the print setting menus, as a number of related Youtube videos never mentioned a problem with it appearing after installing the plugin.
Thanks for your information regarding the log file, which I’ll provide should I find other quirks or bugs that I’ve been able to work around.
I’m really enjoying the updated formats and improvements in these latest versions of Cura.
Regards
Hans König

Ghostkeeper, 2020-04-28:
Closing this because it concerns a different project: https://github.com/fieldOfView/Cura-ZOffsetPlugin

Glad that you found a workaround though. Mind that attachments to e-mail replies to Github don't work. They don't get shared with us. If you intended to upload the cura.log file, please do so on the Github website.

closed 2020-04-28 by Ghostkeeper