Ultimaker/Cura #7576: Include the Z height along with the Layer # in the "preview-slider" text box.

GregValiant, 2020-04-23:
My printer displays the Z height.  Cura displays the layer number.  When editing Gcode it would be nice to have the Z height displayed along with the Layer number in the little text box at the top of the preview slider.  It would also be nice if the layer number in Cura was the same as the layer number in the Gcode file that Cura generates.  Right now Gcode layers start at "0" and Cura layers start at "1".

mahtDFR, 2020-04-28:
Nice idea. We don't have time to do it though, so it's deferred.

louiswouters, 2020-06-08:
@GregValiant I wrote a couple of PostProcessing plugins that will help you. Get them [here](https://github.com/louiswouters/3D-Printing/tree/master/scripts).
One script increase the layer numbers in the G-Code file by 1 so that they match up with the layers in cura.
Another adds the layer number to the screen of your printer.

GregValiant, 2020-06-11:
Thank you sir.
I don't see a download link.  Clicking on the file name opens a window to
view the code.  Please advise.

Greg Foresi

mahtDFR, 2020-06-12:
Check out the readme on that repo, it explains what you need to do.

You'll need to drop the scripts in your Cura installation folder, something like `C://Program Files/Ultimaker Cura X.X/plugins/PostProcessingPlugin/scripts`

Restart Cura and the script will be available to use.

GregValiant, 2020-06-12:
Very good.  Now when I get out of my chair the calculator won't fall off my
lap and I shouldn't make any more mistakes when figuring out what layer to
pause at.

Thanks,
Greg

GregValiant, 2020-06-13:
I like the two plugins.  I do have a comment..

With the two plugins installed, When I slice a part and it has 100 layers
Cura shows layers 1 to 100 in preview.  When the Gcode file is created it
has layers 1 thru 100.  When the part is printing, the LCD shows layers 1
up to 100.  When I open that Gcode file in Cura, layer 1 is missing and the
top layer is 101.  Layer 2 is the new layer 1 and it is on the build
plate.  I will make an assumption that the gcode reader in Cura expects to
see a Layer:0 and re-numbers it to 1 for the preview...but there is no 0.

I'm good with it.  I just thought I'd let you know in case you were going
to put these out there for everyone.

Greg "Everything affects everything".Valiant

louiswouters, 2020-06-14:
@GregValiant Interesting. I played around with this a bit and here's my finding. When opening GCode with Cura, the order of the file doesn't seem to matter. So when you would reverse all layers manually. Cura's preview would look normal. However, your 3D printer will start with the top layer, thus printing in midair. I even tried renaming a single layer, ;LAYER:8 -> ;LAYER:18. This would cause the preview to have an empty layer 9, then at layer 19 it will show that the 9th and 19th layer are printed one after the other. The 3D print would be fine though.

GregValiant, 2020-06-14:
I was thinking that since the layer number lines in the Gcode file are
comments, that the printer would ignore them.  It would just read the file
line by line no matter how the layers were named.  Could be Roman numerals
for all it cared.  That's why inverting the file would invert the print.
The printer just starts at the beginning and goes to the end.  Of course
then all extrusions would be retractions as the E values keep getting
smaller.  It would be sucking air while printing in air.  Hey, I could
print an air guitar!

Cura must look at the layer numbers and make up it' own list, then put them
together in the order they appear to belong.  Does that sound right?  As
long as the files print correctly (and they do) then it's fine.

At any rate I'm pleased that I don't have to keep my calculator in my lap
while converting Z height to layer number.  Much thanks.

Greg

louiswouters, 2020-06-14:
Yes, you are completely right

Ghostkeeper, 2020-06-16:
> When the Gcode file is created it
has layers 1 thru 100.

This is where it's going wrong then. The g-code file should start counting from 0. Unless there's a raft involved. Then it starts at a negative number.

I'm not getting the same result as you though. Here's a snippet from the start of a g-code file:

```
...
G1 F1500 E-6.5
;LAYER_COUNT:98
;LAYER:0
M107
M204 S625
M205 X6 Y6
G1 F600 Z2.27
G0 F4285.7 X106.44 Y97.765 Z2.27
M204 S500
M205 X5 Y5
;TYPE:SKIRT
G1 F600 Z0.27
G1 F1500 E0
G1 F1200 X106.977 Y97.25 E0.01323
G1 X107.566 Y96.795 E0.02646
... (continue printing brim)
```

Ghostkeeper, 2020-06-16:
I'd expect Cura's layer view to be irrespective of layer number comments. It should be able to read g-code generated by other slicers too, after all. Not sure if/how that works though; I haven't been involved much in that part of the code.

louiswouters, 2020-06-17:
@Ghostkeeper 

> > When the Gcode file is created it
> > has layers 1 thru 100.
> 
> This is where it's going wrong then. The g-code file should start counting from 0. Unless there's a raft involved. Then it starts at a negative number.
> 
> I'm not getting the same result as you though.
>

That is with the postProcessingPlugin "[StartLayerNumberingAt1](https://github.com/louiswouters/3D-Printing/blob/master/scripts/StartLayerNumberingAt1.py)". 
It adjusts those comments in the g-code file to match the layer counter in Cura's preview
Apparently, Cura relies on these special codewords ";LAYER:X" so it messes up the preview when the Gcode is loaded back into cura.

GregValiant, 2020-06-18:
Hello,
This is in regards to the "StartLayerNumberingAt1" plugin.
I have a model 60mm tall and slice it with .2mm layer height for a total of
300 layers.  There is a skirt.
:
When I slice the model in Cura with the plugin DISabled, Cura preview is
numbered 1 to 300, the skirt is on layer 1 and layer 1 is on the virtual
build plate.  The gcode file generated is numbered 0 to 299.
When I slice the model in Cura with the plugin ENabled, Cura preview is
numbered 1 to 301 and layer 1 is empty.  The skirt layer is "2" and is
resting on the virtual build plate.  The gcode file is numbered 1 to 300.

While playing with the gcode files that I created I was curious and looked
into the Dog file that came with my printer.  It was sliced with Simplyfy3d
and the syntax for the first layer is *; layer 1, Z = 0.450*..  So at least
one other slicer starts layer numbering at 1.
I opened the Dog.gcode file in Cura.
1. The preview slider doesn't change the layer view (all layers show all
the time).
2. The text box on the slider shows "1" no matter where the slider is
positioned.
3. If you put the top slider ball in the middle of its range, you can slide
the bottom ball past it to the top.

This all seemed so simple when I asked for layer numbering to match between
Cura and the Gcode it creates.  Now my head hurts.
On the other hand, sending the layer number to the LCD is nice and works
well.

Regards,
Greg

>
>

Ghostkeeper, 2020-06-19:
> This is in regards to the "StartLayerNumberingAt1" plugin.

There is no such plug-in or post-processing script in Cura.

louiswouters, 2020-06-24:
> > This is in regards to the "StartLayerNumberingAt1" plugin.
> 
> There is no such plug-in or post-processing script in Cura.

It's a postProcessing script I made. [find code here](https://github.com/louiswouters/3D-Printing/blob/master/scripts/StartLayerNumberingAt1.py). All it does is increase every number after ";LAYER:" by 1 but it seems to mess up the preview in Cura if you load the Gcode back in.

jrmoserbaltimore, 2020-11-16:
Indicating the layer number next to the printing % gives users a spatial sense of where the print is.  This is a quick fix:  a really nice and complete approach would immediately move to the correct layer in the Preview window and advance the print simulation to that point, but of course the preview pane isn't tied to the print job and, besides, that would be a heroic effort.

In general, print % has been a rather useless metric versus layer n of x.

reibuehl, 2021-01-21:
Would it be possible to have the mm value just as a popup when hoovering over the layer count display in preview?

mkoic-gh, 2021-04-28:
The Adaptive Layers feature makes it more important than ever to be able to see exactly how z-height in mm corresponds to layer.  I’d love to see this request implemented.

moebis, 2021-05-06:
I'll throw my hat in the ring, we really need this feature in Cura please. It should be a simply calculation for each layer. What height am I at?

hockeyguy820, 2021-05-31:
I was interested in having this feature so I implemented it myself in the Cura Python and QML code. My unofficial patch adds the layer's height in mm to the layer slider, underneath the layer number. The text file of the patch can be downloaded here:
https://drive.google.com/file/d/1jhxR0Qp3Jrdocv4LTiclolvHdFscvVEU/view?usp=sharing

This patch is based on Cura 4.9.1, but I checked and it also applies cleanly to Cura 4.8.0, and probably will on top of other recent versions as well. You can apply it using the "patch" utility as follows, starting from the top level Cura folder which in Windows would be "C:\Program Files\Ultimaker Cura 4.9.1":
patch -p1 < cura_z_height_patch.txt
If you have Cura checked out using GIT, then you can apply the patch as a commit, as follows:
git am -3 cura_z_height_patch.txt

In addition to showing the Z height, you can also edit the height number to move the layer slider to the desired height, rounded up. I tested using rafts and adaptive layers, and everything seems to work properly. I am not a Cura developer so this patch is unofficial. I recommended backing up the 4 affected source files before applying the patch, just in case:
plugins/SimulationView/LayerSlider.qml
plugins/SimulationView/SimulationSliderLabel.qml
plugins/SimulationView/SimulationView.py
plugins/SimulationView/SimulationViewProxy.py

You can also reverse the patch using the -R option of the "patch" utility. If you have trouble applying the patch automatically, it's not very big so it only takes a few minutes to apply the patch manually by copy/pasting sections of the patch file using a text editor.

reibuehl, 2021-05-31:
Please create a pull request for your patch, @hockeyguy820 ! That would be very useful.

reibuehl, 2023-06-29:
I think this is still a valid request and should be implemented.

shultays, 2023-08-26:
This would be very useful. 

If I notice something weird while printing stuff, I want to check what sliced model looks like at that height/for that layer. My printer shows current height, cura slicer shows the layers using numbers, so hard to tell which layer exactly it is

bengalih, 2024-11-12:
Yes I'm still waiting for this.  As others have mentioned this should be a very simple calculation based on the selected layer height.
Orca slicer does this and it is so helpful.

GregValiant, 2024-11-12:
Still waiting since April of 2020.  In addition it has been mentioned in #8088, #8337, #9564, and #9809.

This is a popular request and (I think) a good idea.  It was deferred at the time the suggestion was made.

@HellAholic is there a chance this could be revisited?  It would be a nice addition.  (I've dropped my calculator so many times it's developing a phobia about calculating Z height from the layer count.)

GregValiant, 2026-09-02:
This feature is now included with Cura and so this can be closed.  

[image]

closed 2026-09-02 by GregValiant