Ultimaker/Cura #7577: Skirt not flowing in continuous line.

lococnc, 2020-04-23:
The purpose of the skirt is to purge nozzle and get flowing, but in 4.6.0 the skirt now changes start point for each row instead of printing entire thing in one continuous line. This causes issues and should be put back the way it was before.

smartavionics, 2020-04-23:
https://github.com/Ultimaker/CuraEngine/pull/1231 probably fixes this issue as well.

ashleysommer, 2020-04-24:
I noticed this too, was about to come to make an issue about it. It used to be one continuous line, now on 4.6.0 it stops the line, and moves over one layer and starts a new line. Not a big deal, but less correct than it was before.

rburema, 2020-04-28:
I've only reviewed the code, but I agree with @smartavionics that the PR will probably fix this issue.

jacekkopecky, 2022-07-31:
I have just installed Cura 5.1 and noticed that the skirt is no longer continuous like it used to be. It doesn't even start near the same point. To illustrate: 

[image]

[image]

[image]

It starts higher every time... That's unnecessary travel moves.

rburema, 2022-07-31:
@jacekkopecky Thanks for the notify, but could you please make a new bug report? This was closed a while ago, and it's likely that the cause is completely different.

jacekkopecky, 2022-07-31:
Will do - I'll look into it some more. This issue isn't closed though @rburema , that's why I posted here.

GregValiant, 2023-09-15:
I'll go ahead and close this as it appears there is another issue open on this same topic.

closed 2023-09-15 by GregValiant