Ultimaker/Cura #7578: Ender3 Pro Build Plate Dimesions Incorrect

ConnorSullivan31, 2020-04-23:
**Application version**
(4.6.0)

**Platform**
(Windows 10)

**Printer**
(Creality Ender 3 Pro)

**Reproduction steps**
1. (I updated CURA to 4.6.0 but did not keep old settings intentionally. I added the Ender 3 Pro as my printer now that it was there unlike before where I had to use the Ender 3 profile. The Ender 3 Pro had the default build volume dimensions set as 235x235x250. The correct dimensions for this printer should be 220x220x250.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-04-23:
Thanks, it was a simple fix, so I pushed it to master. You'll see this changed in the next version.

closed 2020-04-23 by rburema

ConnorSullivan31, 2020-04-23:
Awesome, thanks. I also noticed this later on the Ender 3 as well, but I
never got around to submitting the report

rburema, 2020-04-24:
So you mean the (normal) Ender3 _also_ has a 220x220 bp?

ConnorSullivan31, 2020-04-24:
Yes

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Ultimaker/Cura/issues/7578#issuecomment-618903413",
"url": "https://github.com/Ultimaker/Cura/issues/7578#issuecomment-618903413",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Ghostkeeper, 2020-04-28:
Note that the Ender 3 build plate dimensions have been subject to a lot of discussion.
* Originally it was 220x220: https://github.com/Ultimaker/Cura/pull/4316
* This got changed to 235x235: https://github.com/Ultimaker/Cura/pull/4501
* Then back to 220x220: https://github.com/Ultimaker/Cura/pull/5809
* Then back to 235x235: https://github.com/Ultimaker/Cura/commit/726561bf309616107ad874d4e508b3a3ec9e5701

See also a lengthy discussion here: https://github.com/Ultimaker/Cura/issues/6267
Honestly I don't want to go into it again. There is different firmware for this printer and some firmware has different sizes inside than others. People are disagreeing on whether to count the clips or not. Spec sheets from Creality itself has been changing. Firmware from Creality itself has been changing or not limiting the size. Whatever. I think we need to freeze it to 235 which seems to be the prevailing opinion out of these discussions most of the time.