Ultimaker/Cura #7579: "Creality Ender 3 Pro" printer profile in Cura 4.6 is missing print profiles and nozzle size options.

JAL741, 2020-04-23:
**Application version**
Fresh install of the newly released Cura 4.6.0

**Platform**
Windows 10 x64

**Printer**
Creality Ender 3 Pro (with BLtouch, ABL sensor)

**Reproduction steps**
1. Uninstall any previous version of Cura, choosing to remove all printer profiles.
2. Install Cura 4.6  (fresh / default install)
3. Select printer when prompted: "Creality 3D" --> "Ender 3 Pro"
4. Try to use Cura normally.  FAIL - Unusable default settings.
• Observe that there is are no options to select a 'Nozzle Size' when selecting a print material.
• Observe that Cura reports "unsupported" on the print settings.
• Observe that there is no ability to choose the print layer heigh.
• Observe that there are no options for print profiles.

**Screenshot(s)**
https://drive.google.com/open?id=1Pd0gHFUZTrHhnc3Du-FBtUOAz9Qb60vk

**Actual results**
• Observe that there is are no options to select a 'Nozzle Size' when selecting a print material.
• Observe that Cura reports "unsupported" on the print settings.
• Observe that there is no ability to choose the print layer heigh.
• Observe that there are no options for print profiles.

**Expected results**
• Should be able to select nozzle size, layer height, and other normal slicer options, and slice the STL for printing.  (printer profile "Ender 3" works fine, but printer profile "Ender 3 Pro" does not)

fucena69, 2020-04-23:
Same with me .. need to reistall 4.5 :(

JAL741, 2020-04-23:
@fucena69 there's no need to roll-back to a previous version; just use the "Ender 3" printer profile instead of the "Ender 3 Pro" printer profile; it works just fine.

thekylemars, 2020-04-24:
Same issue here. Ender 3 pro. Oddly I added this printer, becuase all of them disappeared overnight. Printed once fine. Closed. Sometime later opened and noticed no nozzle size. Not even the field to enter it under machine settings. Have 2 other brand printers and they are fine. Profiles I added and default are missing as well on ender 3 pro.

thekylemars, 2020-04-24:
Also missing from ender 3 profile (non-pro). Tried to add that and same result. Going to pull my 4.5 machine in for now.

dexhek, 2020-04-24:
@thekylemars  delete the Ender 3, exit from cura, then open and re-Add the printer, should work..

thekylemars, 2020-04-24:
@dexhek Yup.

venounan, 2020-04-24:
> 
> 
> Same with me .. need to reistall 4.5 :(

It seems that 4.5 doesn't have an Ender 3 Pro profile at ALL. So using the Ender 3 on 4.6 seems the same as using 4.5. But 4.6 was my first version and it not having pro profiles was a bit confusing for me. Will just use the 3 normal profiles in 4.6 for now and hopefully we'll get them in the next version.

dexhek, 2020-04-24:
@venounan actually the problem was solved  here #7550

venounan, 2020-04-24:
> 
> 
> @venounan actually the problem was solved here #7550

Perfect, thank you!

diegorafaelss, 2020-04-27:
I had the same problem in Cura 4.6 when selecting the Ender 3 Pro. I went back to the basic Ender 3 profile. I believe it will soon be corrected!

dexhek, 2020-04-27:
@JAL741 i think the issue could be closed, since it has been solved.

Ghostkeeper, 2020-04-28:
Duplicate of #7550.

Ghostkeeper, 2020-04-28:
> Same with me .. need to reistall 4.5 :(

Note that this printer profile is new in 4.6.

closed 2020-04-28 by Ghostkeeper

JAL741, 2020-04-28:
@Ghostkeeper, yes I know that "Ender 3 Pro" is a new profile as of Cura 4.6.0, and no it is not fixed in 4.6.0; that is the version on which this bug was identified.

Ghostkeeper, 2020-04-28:
Patience, man. I was trying to point out that going to 4.5 is not a solution in this case.

It's common in bug trackers to close issues when they are fixed in a development build, rather than when they are released in a stable build. I know you want to get started with the new profile as soon as possible, but it'll have to wait a little bit longer.

GoguyT3D, 2020-06-28:
(I had the same problem) I have read all of your comments and attempted to change my printer from an Ender 3 Pro to and Ender 3, and that fixed it. I have just changed it back to an unnetworked Ender 3 Pro and all of my default print profiles are back. FYI I never lost nozzle size selection during or after this bug. Hope it gets repaired in the next update.

JAL741, 2020-06-28:
It was resolved in version 4.6.1 a few weeks after my original post