Ultimaker/Cura #7581: cura 4.6

neronear, 2020-04-23:
I cannot see anteprima after slicing

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-04-24:
So, 'anteprima' is a translation of 'preview'. Do you mean either of the preview _buttons_, or is the preview _itself_ staying blank somehow? What _exactly_ are you seeing?

A screenshot does wonders in these situations.

Ghostkeeper, 2020-04-28:
Please fill in the issue template next time.

closed 2020-04-28 by Ghostkeeper