Ultimaker/Cura #7582: [Cura 4.6] Inconsistent percentage appearing during Scale

hpandelo, 2020-04-23:
**Application version**
4.6.0-Beta

**Platform**
OS. Windows 10 1909 - Build: 18363.752
CPU. AMD Ryzen 7 3700X
GPU. AMD Radeon RX 5700

**Printer**
Ender 3 PRO (Custom GCode to Beep when starting and finishing the print)

**Reproduction steps**
1. Select the model
2. Click on Scale Menu (or hit S on keyboard)
3. Uncheck all checkboxes to make sure nothing is acting into your values
4. Resize X (by percentage or value)
5. Rezise Y (by percentage or value)
6. Observe value and percentage of Z

**Screenshot(s)**
Attaching a GIF
[image]

**Screenshot(s) - EXTRA:**
When I hit "Reset" the value is replaced with the original, but not Y percentage. 
_PS* This isn't happening always!_
[image]

**Actual results**
Percentage of Z is getting affected by changes in Y field

**Expected results**
Value and Percentage of Z (or any other) should **not** be affected by changes into unrelated fields

**Project file**
Any

rburema, 2020-04-24:
Have you rotated the object before scaling? (I ask this because if so, then it's probably a known issue.)

hpandelo, 2020-04-24:
@rburema Yeah.. I did it. The original model is placed in Vertical, I rotate to be Horizontal

Andy1978, 2020-04-25:
I guess it's this one: https://github.com/Ultimaker/Cura/issues/4136

dxgldotorg, 2020-04-27:
@rburema Running into scaling issues with rotated parts too.

Ghostkeeper, 2020-04-28:
Yup, duplicate of https://github.com/Ultimaker/Cura/issues/4136. Sorry.

closed 2020-04-28 by Ghostkeeper