Ultimaker/Cura #7583: [Pause at Height] Custom GCode option

rodrigosclosa, 2020-04-23:
Hi,

I would suggest to implement a custom GCODE field on the Pause at Height plugin to put some custom code before Pause occurs, for example, to beep some tunes to alert the user it's paused.

I'm a programmer, I can do it. How can I help with it?

Tks,
Sclosa.

rburema, 2020-04-24:
You might want to look at [this pull request](https://github.com/Ultimaker/Cura/pull/6778) that was recently merged into master for inspiration on how to proceed.

In general, the post processing scripts are relatively easy do develop for (compared to other parts of Cura), you won't need to build all of Cura, just add to the postprocessing/script folder (or even put your script module in the %appdata% scripts folder).

Ghostkeeper, 2020-04-28:
Let's continue discussion here at the pull request: https://github.com/Ultimaker/Cura/pull/7585

closed 2020-04-28 by Ghostkeeper