Ultimaker/Cura #7584: [4.3, 4.4.1, 4.5, 4.6] Cura crashes on startup

mbeckham, 2020-04-23:
**Application version**
Tried on 4.6, 4.5, 4.4.1 and 4.3 before giving up

**Platform**
OS: Windows 10
CPU: i7-9700
GPU: RTX 2060 super
RAM: 32GB

**Printer**
none, cura wouldnt start

**Reproduction steps**
1. install cura
2. attempt to run cura

**Actual results**
Splash screen appears and appears to be loading, main screen flashes but immediately closes, icon in tray also dissapears

**Expected results**
Cura should startup successfully

**Log file**
Uploading the cura.log file and the stderr files from %APPDATA%\cura\stderr.txt as they look to be more helpful, pasting a snippet from the end of the 4.6 stderr file here:

```
20-04-23 17:38:09,832 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-04-23 17:38:09,833 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [39]: No active GlobalStack, set CustomQualityProfilesDropDownMenuModel as empty.
2020-04-23 17:38:10,047 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [207]: Support for Vertex Array Objects: True
2020-04-23 17:38:10,060 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2020-04-23 17:38:10,062 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 441.41
2020-04-23 17:38:10,063 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2020-04-23 17:38:10,065 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: GeForce RTX 2060 SUPER/PCIe/SSE2
2020-04-23 17:38:10,066 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
2020-04-23 17:38:10,068 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-04-23 17:38:10,074 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\selection.shader]...
2020-04-23 17:38:10,080 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\select_face.shader]...
2020-04-23 17:38:10,085 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\default.shader]...
2020-04-23 17:38:10,088 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\composite.shader]...
2020-04-23 17:38:10,095 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\overhang.shader]...
2020-04-23 17:38:10,101 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
2020-04-23 17:38:10,107 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\transparent_object.shader]...
2020-04-23 17:38:10,113 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\striped.shader]...
2020-04-23 17:38:10,115 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
2020-04-23 17:38:10,121 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray_composite.shader]...
2020-04-23 17:38:10,127 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\platform.shader]...
2020-04-23 17:38:10,135 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
Fatal Python error: Aborted

Thread 0x00003604 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 78 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003500 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.6\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002ddc (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003650 (most recent call first):
  File "threading.py", line 297 in wait
[... 91 lines cut ...]
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00001d68 (most recent call first):
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.6-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 813 in run
  File "X:\4.6-exe\build\inst\bin\cura_app.py", line 231 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

```
**Additional information**
Attempted to run as administrator on all versions with no difference in behavior

[cura.log](https://github.com/Ultimaker/Cura/files/4525691/cura.log)
[stderr4.3.txt](https://github.com/Ultimaker/Cura/files/4525692/stderr4.3.txt)
[stderr4.6.log](https://github.com/Ultimaker/Cura/files/4525693/stderr4.6.log)

rburema, 2020-04-24:
> UM.View.GL.OpenGL.\_\_init__ [112]: Initialized OpenGL subsystems.
 UM.View.GL.OpenGL.\_\_init__ [113]: **OpenGL Version:  4.1.0** NVIDIA 441.41
UM.View.GL.OpenGL.\_\_init__ [114]: OpenGL Vendor:   NVIDIA Corporation
UM.View.GL.OpenGL.\_\_init__ [115]: OpenGL Renderer: GeForce RTX 2060 SUPER/PCIe/SSE2
UM.View.GL.OpenGL.\_\_init__ [116]: **GLSL Version:    4.0.0**

Something strange is going on with your graphics card, or drivers thereof. We ask for a OpenGL 4.1 context, which _should always_ support glsl version 4.1 (see [here](https://en.wikipedia.org/wiki/OpenGL_Shading_Language#Versions)), and indeed our shaders require `#version 410` of glsl.

But it seems your setup reports glsl *4.0* somehow, despite running OpenGL *4.1*. I suspect the drivers may be borked somehow, perhaps they should be updated or reinstalled.

mbeckham, 2020-04-24:
Yep, uninstalling and reinstalling my graphics driver worked! thanks!

closed 2020-04-24 by mbeckham

justin13888, 2020-05-10:
I had the exact same issue. I had to uninstall the whole driver and then reinstall to fix it. It must be some Nvidia driver problems with Cura. It works but it still says it's using glsl 4.0, not 4.1.

mbeckham, 2020-05-11:
> I had the exact same issue. I had to uninstall the whole driver and then reinstall to fix it. It must be some Nvidia driver problems with Cura. It works but it still says it's using glsl 4.0, not 4.1.

Yea, mine worked for a day or two after reinstalling and then went back to doing the same error, not exactly sure what is causing it to change, but something definitely did as it went from working to reverting to the listed behavior.

I've just accepted that I need to do all of my cura work on my linux partition for now.

justin13888, 2020-05-11:
Mine broke a few minutes after working but it somehow works again now. I don't think this is a driver issue but more certainly, a Cura bug with Nvidia graphics cards for Windows. Hopefully, they figure out the issue.

timkoers, 2020-07-02:
I'm not sure if you have an Asus GPU, but enabling the OSD overlay of GPU drivers can cause issues.

justin13888, 2020-07-12:
Yes, I have an Asus GPU.

> I'm not sure if you have an Asus GPU, but enabling the OSD overlay of GPU drivers can cause issues.

It seems to have solved this problem for me if I disable OSD. Thanks.

MillerPesh, 2021-01-03:
> 
> 
> I'm not sure if you have an Asus GPU, but enabling the OSD overlay of GPU drivers can cause issues.

THIS is still relevant in January 2021. If you have ASUS GPU, the OSD feature in GPU tweak is likely causing the problem for you. Exiting GPU tweak fixed it for me. Thanks for the tip stranger.

Colmanian, 2021-05-14:
Crashes on startup after upgrading to 4.9.1

rburema, 2021-05-18:
@Colmanian did any of the fixes in this thread work for you? Alternatively, it might be something unrelated to these problems, could you share the logs?

windowslucker1121, 2021-12-31:
> eems to have solved this problem for me if I disable OSD. Thanks.

the problem still exists in dec 2021.
running a rog strix 3080 with asus gpu tweak 3 and enabled OSD.
temp fix is to disabled the OSD.
thanks for your advice

[cura.log](https://github.com/Ultimaker/Cura/files/7796488/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/7796489/stderr.log)

Ghostkeeper, 2022-01-11:
I think, if Mbeckham's solution didn't work for you, you probably have a different problem.

windowslucker1121, 2022-01-11:
like mentioned in my post
`temp fix is to disabled the OSD.`
i got it working, i just wanted to point out, that its still not fixed