Ultimaker/Cura #7587: The first blob of filament at the start of each print hits the bottom the cartridge and fan shield

christopherwestcott, 2020-04-24:
Ultimaker 3 Extended
Cura 4.6 (but the same problem has happened on all versions)
Window 10

Problem:
At the beginning of every print, the printer goes to the front left corner (closest to the door) and deposits a blob of filament, then moves at a downward diagonal movement towards the back. The blob is tall enough to smear onto the nozzle and the cartridge. It is not very clean. At the end of the print job, I always have to clean up the nozzle.

What I would expect it to do after depositing a blob is to lift up a little and straight back before moving in its normal path, to keep the nozzle clean.

rburema, 2020-04-28:
A workaround is to use skirt instead of a prime-blob. It turns out skirt is enough to prime the filament in many cases.

Ghostkeeper, 2020-04-28:
Actually this movement is not implemented in Cura or encoded in the g-code, but implemented by the Ultimaker 3 firmware. There is not really something that we can do about it. I've seen it happen as well sometimes.

Ghostkeeper, 2020-04-28:
It seems that the Ultimaker firmware team knows about this issue (EM-2157). It's been on their backlog since 2018-2.

closed 2020-04-28 by Ghostkeeper