Ultimaker/Cura #7588: v4.6 creates significantly more gcode than v4.5, resulting in poor printing performance

ilektron, 2020-04-24:
**Application version**
4.6.0

**Platform**
Mac OS X, Macbook Pro Intel i9 9900k with Radeon Pro Vega 20 4 GB

**Printer**
Creality Ender-3

Cura is printing to Octoprint running on a Raspberry Pi 3B+. Printer is actually an Ender 3 Pro running klipper firmware.

**Reproduction steps**
For the screenshots below, 
1. On Cura 4.5.0, load test model with curves
2. Export model as gcode
3. Update to Cura 4.6.0 
4. Load model in 4.6.0
5. Export gcode 
6. Compare exported files

**Screenshot(s)**
Top is 4.5.0 and bottom is 4.6.0
[image]

**Actual results**
The gcode has increased over 20% on a 3DBenchy scaled to 75% from 57k lines to 70k lines and the file size when from 1.7MB to 2.1MB. The issue seems to affect curves, and isn't readily visible on straight lines.

The increased gcode size is causing stuttering when using Octoprint with a Raspberry Pi 3B+ connected to an Ender 3 Pro. Originally, I assumed it was the change in the default Maximum Deviation under mesh fixes from 0.05 to 0.025, but even when changing this back in 4.6.0, the gcode is significantly larger

**Expected results**
I would expect similar performance between versions and comparable gcode output.

**Project file**
[Gcode.zip](https://github.com/Ultimaker/Cura/files/4531596/Gcode.zip)

**Log file**
(nothing worth noting)

**Additional information**
I've attached the gcode for comparison as a result. There are known performance issues with Octoprint and high detail models, as well as with curves and circles, however the change in gcode generation has made 4.6.0 unusable for my setup.

smartavionics, 2020-04-25:
I took a quick look at your gcode, it seems to have a lot of tiny line segments which are causing the quality issues you are seeing. I think the regression cited in https://github.com/Ultimaker/Cura/issues/7557 is also responsible for your problem.

neophrema, 2020-04-27:
Not sure what you are suggesting. The new update rendered my configuration unusable aswell.
Im using Raspi/DWC2/Klipper - the Printer is stuttering very much and I get poor visual results also.
Is this a configuration issue, is this a bug/false implementation?

smartavionics, 2020-04-27:
I think there is a regression in 4.6.0 that is causing multiple quality issues...

neophrema, 2020-04-27:
So the stuttering might get a fix? Did i understand correctly, that the Master branch is fixed already?

smartavionics, 2020-04-27:
Well, it may be fixed already but I can't verify that because I don't use the master branch directly.

rburema, 2020-04-28:
If this is indeed that issue, we'll be coming out with a patch (4.6.1) this week.

Ghostkeeper, 2020-04-28:
> So the stuttering might get a fix? Did i understand correctly, that the Master branch is fixed already?

The master branch has the algorithm reworked for like 50%. We've tested that with the Cura team and it was deemed to work like a charm, though word from Ultimaker's broader quality assurance team is still out.

In 4.6 we cherry-picked one of the changes from that rework, but it seems that one broke a couple of things when taken separately from the rest of the changes. We're reverting that with 4.6.1.

Ghostkeeper, 2020-06-04:
The resolution settings and compensation for wall overlap is the same between those two printers, so I guess you just have general printing problems then. Needs tuning of the profiles.

Ghostkeeper, 2020-11-03:
With the changes to the Maximum Resolution/Deviation algorithm in 4.6.2, 4.7.0 and 4.8, any results from this issue will have changed radically. See https://github.com/Ultimaker/Cura/issues/8321 for more recent discussions.

closed 2020-11-03 by Ghostkeeper