Ultimaker/Cura #7589: Move user settings to Documents folder

hpandelo, 2020-04-25:
**Is your feature request related to a problem? Please describe.**
I've formatted my disk recently and with this change, I wouldn't be able to backup the profile settings as a common user can do. I did it as copy/paste the data from the old `%AppData%` folder, but a common/normal windows user probably won't know about how to do it.

**Describe the solution you'd like**
Instead of `%AppData%/cura/<version>/....` use `%HomePath%/Documents/Cura` for all custom stuffs 
```ruby
E.g.: 
      My Printer machine settings (not all available, just the printers I've added)
      Materials added by me
      And Custom Profiles
```

fieldOfView, 2020-04-25:
`%AppData%/cura/<version>/....` is what Microsoft prescribes.

nallath, 2020-04-28:
Not only that, we actually provide a backup system from within Cura itself (which allows you to store 5 backups)

closed 2020-04-28 by nallath