Ultimaker/Cura #7590: v4.6 prints support structure even when it is switched off

g-druckt, 2020-04-25:
**Application version**
v4.6

**Platform**
Win10 Pro, Intel i7, NVidia GPU

**Printer**
Anycubic Mega-S

**Reproduction steps**
If the checkbox 'tree structure' ("Baumstruktur" in German) is checked, cura is printing a support structure even when support structure is swichted off in the settings. This behaviour is different to previous Cura versions. Although it is displayed correctly in the preview (i.e. with support), this must be considered as a bug, because it's not what you expect Cura to do. The dependency to the tree structure could only be found by trial and error.

**Screenshot(s)**
n/a 

**Actual results**
Model is printed with support structure although this is not wanted.

**Expected results**
Well, I think that's obvious ...  ;-)

**Project file**
https://www.thingiverse.com/thing:4278873 (--> medium sized mask)

**Log file**
not available

**Additional information**
(none)

discip, 2020-04-25:
@g-druckt
Good morning sir,
I think I know why your result is different than your expectation.
**But I do not know for sure!**

1. Could it be, that you switched off the *"helpers"* in *"Line Type"*?
[image]

2. If *"tree support"* is ticked it **always generates support!** Whether *"generate support"* is activated **or not**.
In fact if you tick **both**, there will be generated **2 kinds of support**:
a. The *usual one* and
b. the *"tree support"* intersecting each other **even in the previous builds!**

Please try the above mentioned settings.
kind regards

g-druckt, 2020-04-25:
Regarding 1) I don't know where to find this setting (so I defintely havn't changed it). But the preview is not my issue anyway. I have seen the support structure in the preview, but (at first) I couldn't find a way to switch it off.

From my experience your statement about the tree support (topic 2) is not correct. I have always had tree support enabled and switched on or off support generation with the 'generate support" checkbox. Having tree support enabled (or not) only changed the way the support structure was looking like (provided that "genarete support" was enabled). Expecially when using the simplified settings I find it very confusing to have support generated even when it seems to be swichted off (you don't even see the "tree support" checkbox then). I was using v4.4 before and I am very sure the behaviour was different there.
  
In summary I can live with the existing implementation, but I still consider it to be not correct.

rburema, 2020-04-28:
@g-druckt Tree support has _always_ been independent of normal support, this was never changed! Arguably this is a problem, but the behaviour has not changed. In fact, that's the thing that would most likey prevent us from 'fixing' this now (because the simple fix _is_ a change). Eventually we'll move tree-support out of experimental probably, and make support a drop down of none. normal, tree, but that'll take a while (there are still one or two issues in Tree Support that prevent it from being not-experimental).

@discip Those are visualisation settings, and even if they weren't, the problem is the other way around, something _is_ there, when the user thinks there _shouldn't_.

Ghostkeeper, 2020-04-28:
Maybe this problem really originates from a bad German translation.

The English original is "Tree Support".
The translation for "Enable Support" is "Stützstruktur aktivieren".
Ergo, the translation for "Tree Support" should be "Baumstützstruktur" rather than just "Baumstruktur".

I'm not a native German speaker though, just from what I learned at school and out of my hobby of subtitling.

Ghostkeeper, 2020-04-28:
We have an open ticket to merge normal support with tree support. I think the most appropriate solution would be to have a drop-down to choose whether you want area support or tree support, which is a setting under the normal support category and gets visible only when Enable Support is checked.

g-druckt, 2020-04-28:
@Ghostkeeper : Yes, I agree with both your comments. The German translation really is a little misleading and your proposed solution sounds good to me.  :-)
The only thing which is a little strange to me is why I didn't have the support printed those few times when a wanted to print without it (because as already written, I only turned off "Enable Support" and had "Tree Support" enabled all the time ... and I'm sure about this). But I don't want to install v4.4 again and try to find out what was going on there.
As you already have an open ticket it's ok for me to close this topic here. Thank you for your support!

Ghostkeeper, 2020-04-29:
The "open ticket" I'm referring to is only a ticket in our internal tracking system. We can keep this one open too to prevent duplicates from being reported.

I'll correct the German translation for the next release, but we might need to also adjust some more tree support related settings.

Ghostkeeper, 2020-04-29:
Nevermind that. It seems that the rest of the tree support settings already properly call it "Baumstützstruktur".

rburema, 2020-09-08:
For 4.7, tree-support has been moved out of 'experimental', and the choice for 'support' is now a dropbox that offers _either_ 'normal' or 'tree', but not both. Considered fixed, then!

closed 2020-09-08 by rburema