Ultimaker/Cura #7591: Cura 4.3.0 - Support Roof Line Directions not showing

Anar91, 2020-04-25:
**Application version**
Cura 4.3.0

**Platform**
Windows 10

**Printer**
Creality CR10S Pro

**Reproduction steps**
1. Turn on "Support Interface Line Directions" (checked)
2. Turn on "Support Roof Line Directions" (checked)

**Screenshot(s)**

**Actual results**
Only "Support Interface Line Directions" is showing.
Therefore "Support Roof Line Directions" can't be changed.
Also changing "Support Interface Line Directions" seems to do nothing.

**Expected results**
"Support Roof Line Directions" should show so it can be modified.
Modifying "Support Interface Line Directions" should also work as expected.

**Project file**

**Log file**

**Additional information**
Thanks for making Cura such a complete slicer !

rburema, 2020-04-28:
Maybe I'm misunderstanding your problem, but if not: Those settings are both comma separated lists of integers, not checkboxes.

Otherwise: Does this happen in the latest version (4.6) as well? We're not going to update 4.3

Ghostkeeper, 2020-04-28:
I think he means to check them in the setting visibility screen.

Ghostkeeper, 2020-04-28:
I'm not able to reproduce this bug. Are you sure that "Enable Support Roof" is activated? The setting will not show up if you have no support roof to change the line directions of.

closed 2020-06-19 by no-response[bot]