Ultimaker/Cura #7592: Discard or Keep Changes Dialog is Confusing (UX Bug)

AntonyCook, 2020-04-25:
1. Modify a profile parameter e.g change Build Plate Adhesion from Skirt to None
2. Switch to a different quality profile

**Screenshot(s)**
[image]

**Actual results**
Dialog pops up asking you to Discard or Keep Changes.

I have raised this as a UX Bug because this it is really confusing as to what Keep or Discard will actually do. 
Every time this window pops open I wonder the following things
1) what is Default ? If I'm moving away from a Profile that I have customised and saved in the past, 'Default' sounds wrong because it's not a system wide/cura default, it's actually a customised value I've saved to that profile in the past....it's not a default, it's my personalised setting..does default perhaps pertain to the default in the new profile I'm switching to ?
2) what is customised ? is it the setting in the profile I'm about to change to  (since most times I'm switching to one of my own customised profiles)
3) what on earth is Keep going to do ? Is it going to retain the current value in the current profile and copy it across to the new profile ? Or is it just complaining I have customised, my already customised profile and have not saved the profile...in which case, is cura going to save my customised profile for me and then switch to the new profile, or is it going to keep this customised value and apply it to the new profile I'm switching to
4) I'm really confused by this point...what will Discard do ? 

**Expected results**
I think it would be much clearer if
1) the box dialog was titled "Save Changes to <profile you're moving from>?"
2) Default was renamed to "Saved Value"
3) Customized was renamed to "Current Value"

Keep was changed to "Update <profilename> Before Switching Profiles"
Discard was changed to "Switch profiles without updating <profilename>"

...OR something along those lines !

Ghostkeeper, 2020-04-28:
I'll answer the questions directly for your satisfaction, but that doesn't solve the UX problem in general of course but it does change your expected results / proposed solution I think:

> what is Default ? If I'm moving away from a Profile that I have customised and saved in the past, 'Default' sounds wrong because it's not a system wide/cura default, it's actually a customised value [...]

Indeed "default" refers to the profile value of the profile you're switching to. You've changed the adhesion type and didn't save it to a profile. You could revert that change to the profile's default or you can keep your unsaved override.

> what is customised ? is it the setting in the profile I'm about to change to (since most times I'm switching to one of my own customised profiles)

No, it's the current value that you had before you switched profiles. You've made a change to this setting that is not saved to a profile. That change is shown here.

> what on earth is Keep going to do ? Is it going to retain the current value in the current profile and copy it across to the new profile ? Or is it just complaining I have customised, my already [...]

"Keep" will keep your customised value, so `none` in this case.

> I'm really confused by this point...what will Discard do ?

"Discard" will remove your custom override and instead use the default of the new profile.

The intention of the dialog is not to allow you to save the current changes to a new profile. The intention is to ask you whether you want to keep your unsaved modifications and apply them on top of the profile you're switching to, or whether you want to discard those modifications and instead work from the defaults of the profile you're switching to.
The dialogue makes no distinction on whether you're switching to one of the built-in profiles or a profile that you saved yourself a month ago.

AntonyCook, 2020-04-29:
That's really helpful and completly opposite to what I thought then ! So to summarise, in most applications when you move away from an unsaved changed, the application typically prompts you to save the current context, before moving to the new context but in Cura, it prompts you to merge (Keep) the differences into the new context. That's the crux of my confusion and unfortunately, a more fundamental change I guess. Hopefully this will generate some discussion to see what the community thinks as to a better way forward.

nallath, 2020-04-29:
We've at least made a small bit of this more clear:
[image]
(it will now mention the profile you are switching to instead of always saying 'default')

AntonyCook, 2020-04-29:
thanks ! that's a step in the right direction I think. I wonder if in addition, you might consider rephrasing the second line of the summary text in this case to read...

"Would you like to Keep these settings after switching profiles, or load them from _'Fast'_ (Discard)?"

I think that would go a long way to making it clear in advance of anything more involved.

rburema, 2020-04-29:
Done! Had to split the question in two lines to leave some space for the profile name though.
It now says:
_"You have customized some profile settings.\nWould you like to Keep these changed settings after switching profiles?\nAlternatively, you can Discard the changes to load the defaults from '%1'."_

kaeverens, 2020-07-14:
have to agree with this issue. every time I see this poup (I'm looking at one right now) I wonder what is changing.

the text in the version I'm looking at says "Would you like to keep or discard those settings". Well, I made those changes to the old profile for a reason, so I would like that old profile to keep the changes that I have made. But now, the comments above suggest that it has nothing to do with keeping/discarding changes to the old profile, but has to do with pulling those changes over to the profile I'm moving to?

Confusing.

rburema, 2020-07-15:
@kaeverens This change was already made, so it's going to be in 4.7

Concerning 'copy' instead of 'keep' though, I think that could be read as something that's altered in the profile _permanently_, rather than just on top of it.

Not sure if I had an actual ticket for it back then, or if the change was considered small enough to do so without, so I'm closing this ticket now (since, like I said, this will be addressed in the upcoming 4.7).

closed 2020-07-15 by rburema

MisterScott, 2020-09-01:
It seems to me that the developers (who know the program internals) have a clear understanding of what "keep" means, but users don't have that background knowledge.  When switching from profile X to profile Y, in a comment above we see '"Keep" will keep your customised value'.  What we don't know is what exactly "keep" means.  **Where** and **how** **long,** exactly, will it 'keep' it?  

Please clarify.  When switching from profile X to profile Y, does "Keep" modify:

- profile X (switching AWAY from)?
- profile Y (switching TO)?
- an override on top of the loaded profile, which will apply only to the current session unless you save the profile?
- some combination of these?
- something else?

rburema, 2020-09-02:
@MisterScott  Re:Keep: The changes will be kept in the sense that they are transferred over to the 'switched to' profile. There, they'll still register as changed values (the little 'restore to profile' curved arrow will be next to them), so the changes aren't saved permanently to the 'switched to' profile, but they do, like all changed values, persist throughout a restart.

Conceptually, this is closest to your third option, but it's slightly smarter in the sense that the changes persist across sessions (unless of course you reset/discard them in some way). I guess you could say that makes it a sort of 4th option 'smart compromise' between the 2nd and 3rd options?

Re:Discard: Yes, the defaults will be chosen instead (you can see which ones will be replaced in the table).

AntonyCook, 2020-09-02:
Is there really a (user) need for this complexity? I get that it's quite flexible and might be some outlying cases where it would be helpful but I'm sure the vast majority of users just want Keep = save the profile changes and move on, Discard = drop the unsaved changes and move on.

At the moment we have 

Keep = Don't save the random changes you've made to the current profile, just apply them to the new profile....don't save them there either, but when you restart Cura, they'll still be there"

I think the only option I need is Discard, the connotations of Keep are mind-boggling !

rburema, 2020-09-02:
@AntonyCook Except that everyone _starts_ their comment the same way you do (the complexity of this screen is a bit much), but then has a _completely_ different idea of what to throw out and what to keep (pun not intended). Some people will always want to 'Discard', some will need exactly what 'Keep' does now, etc.

'Keep = Save' will not be as clean as you think it is. It'll mean we either _overwrite_ the base profile (a complete nightmare to give users any kind of support on), or saving to a different profile (based of the one before) _every_ time someone switches and has made changes. That'll be a _very_ long list for some users.

In any case, with a sufficiently large and diverse user base, some sacrifices will have to be made.

AntonyCook, 2020-09-03:
I appreciate the dilemma ;-) user research is the only way out...

AntonyCook, 2020-09-03:
just as an aside, I was looking for documentation on how the profiles work and found this page...I think it would be useful to update the wording to clarify...

https://support.ultimaker.com/hc/en-us/articles/360012381720-Switching-print-profiles-with-unsaved-changes

so that Point 1 and 3 read..

Keep (4). Keep the unsaved changes, they will be injected into the ‘fine’ profile, resulting in a ‘fine profile’ with the same modifications as performed in the previous profile. **Note: The current profile (being switched 'from') is neither updated nor saved when Keep is selected - the unsaved changes are effectively 'moved' to the 'fine' profile to be retained, and are now unsaved changes in that profile instead.**

Create new profile (3). The displayed settings will be saved as a new profile, and the ‘fine profile’ will load as the protected, unchanged profile. **The current profile (being switched 'from') will remain unchanged.**

Ghostkeeper, 2020-09-03:
> Is there really a (user) need for this complexity?

There is a preference for this that allows you to tell Cura to always keep the changes. Or to always discard them.
There are two workflows that we found we needed to support here.

Some people make changes to the settings that they want to apply to a multitude of prints because they think they are more or less always better, but they don't want to make a profile for it per sé (because hassle). Those people will want to keep the changes. Myself included.

Some people make more inherent changes to the profiles that need to play well with the current profile. For instance they find that the heightened printing temperature means that you need to retract faster as well. If you then switch to a different material that's printed with a lower temperature, they want the retraction speed to reset. They really intended it to work only for that print or series of prints.

In neither case is the current profile changed. In many cases the current profile can't be changed, because it's a built-in profile.

AntonyCook, 2020-09-04:
thanks Ghostkeeper, having read the other threads mentioned above I appreciate there are different workflows and clearly end users take advantage of features and build them into their day to day process. Also appreciate how well the team are handling some less than constructive feedback from some in the community. So happy to concede that the complexity is an asset and serves to address a wide range of user journeys. 

People are genuinely confused/frustrated by the UX tho I think and these threads will keep reoccurring so my suggestion would be to publish an in-depth guide on how the profile system works/is intended to be used - showcase the power of Curas profiles and how it is a differentiator....diagrams would be good if they would serve to explain how the different types of profiles interact (I must admit, I'm not 100% clear whether when I save a profile, every one of those options is saved to the profile, or whether some of them relate to the material I've selected and get saved elsewhere). A top-down, summary of the profile system and its concepts, with some scenario-based descriptions would go a long way to avoiding confusion (or at least give UM something to refer people to when these threads are created).

I still think the choice of words on the screen is confusing, but maybe it wouldn't be so confusing if I understood how the profile system worked...

If you want to contact me directly, I'm more than happy to help review a wiki page or whatever from the perspective of a newbie....and If I can understand it from that, I'm sure anyone could!

AntonyCook, 2020-09-04:
ahhh ! thanks, will take a read through and see If I can follow those in relation to the app itself. I haven't updated to 4.7 yet (worried about losing all my configuration!) but will do this weekend.

nallath, 2020-09-04:
And some more info can be found on https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

MisterScott, 2020-09-04:
Perhaps what is needed is graphics.  
Show a box for previous profile settings (maybe with a Save button added?  What if I wanted to save the changes?).  Box for new profile.  Wide arrows from the profiles going through a box showing the changes / overrides it's asking "keep/discard" about.  Arrow then points to the right or to a caption "Settings Used" or similar.

Gilligan8, 2025-06-06:
Why not state it as "Keep Current Settings" and "Load Profile Defaults".

I too am constantly going, "wait... is it saying it will discard my changes or it will discard the changes that the profile will make that is different than mine?"

I feel like the buttons could be much more easily spelled out and I don't understand why that as far as I know (version 5.3 to 5.10 this is still the dialog box. :(

Full discloser, I just took the time to read the text at the top of the box and it does spell it out better... but changing those buttons feels like a much quicker way to spell it out.