Ultimaker/Cura #7594: Custom support cause line on model

keremdikmen, 2020-04-25:
**Application version 4.5.0**
(The version of the application this issue occurs with.)

**Platform**
Windows 10, GPU= RX570 Ram= 16GB Processor= AMD FX-8350

**Printer**
My own build CoreXY Printer

**Reproduction steps**
1. I tried to make custom support with support blocker.
2. After i did this. I used "Per Model Settings" > "Modify settings for overlaps" section and check Generate support.

**Screenshot(s)**
1) https://yadi.sk/i/QGDbWXHTQY1E-Q
2) https://yadi.sk/i/OJjI9DyfDhM7hw

**Actual results**
Supports are generated but all the intersection has structure lines.

**Expected results**
It must just generate supports and there shouldn't be any line

**Project file**
[Gcode](https://yadi.sk/d/vez3cpgijj6pNQ)

fieldOfView, 2020-04-26:
Instead of "Modify settings for overlaps" try using "Print as support".

"Modify settings for overlaps" internally creates a new mesh out of the union of the modifier and the mesh that it overlaps. This new mesh has its own walls, so you get a seam between the walls of the original mesh and the walls of the new mesh. You can see whats happening in the layer view.

Ghostkeeper, 2020-04-28:
Correct. Working as intended there.

closed 2020-04-28 by Ghostkeeper