Ultimaker/Cura #7595: Leyer Height Bug

marco-117, 2020-04-25:
**Application version**
4.6.0

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. Slice a Benchy Model with Lexer height  of 0.2mm

**Actual results**
The printed Steps are 0.16mm

rburema, 2020-04-28:
Can you attach a project file to the issue? (Save, don't export. Rename or zip to appease github.)

Ghostkeeper, 2020-04-28:
I'm not able to reproduce this issue so far. Here is a snippet of g-code from the start of the print, where you can see it move to Z0.2 which is the initial layer height:

```
;LAYER_COUNT:50
;LAYER:0
M107
G0 F6000 X122.637 Y102.311 Z0.2
;TYPE:SKIRT
G1 F2700 E0
```

The second bug you're reporting is probably due to the [Skirt/Brim Minimum Length](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/platform_adhesion/skirt_brim_minimal_length.md) setting. Reduce that to 0 if you always want it to have the Skirt Line Count setting regardless of how small the skirt is then.

marco-117, 2020-04-28:
[image]

At first a screenshot. As you can see, the model has a heigt of 48mm and it is sliced into 275 layers. So one layer has a height of 0.17mm and NOT 0.2mm as in the configuration.

I added the project as you want, i hope it will help
[CE3_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/4547900/CE3_3DBenchy.zip)

The explanation with the skirt should be the solution. The model is small and therefore one skirt line would be to short. Thank you for that.

marco-117, 2020-04-28:
Here the same Model and same configuration in Cura 4.5.0. The amount of layers fits to the layer hight in the configuration.

[image]

nallath, 2020-04-28:
You seem to be using the dynamic profile. That has variable layer height enabled. The second profile doesn't use that. So you are simply comparing things that aren't the same.

Ghostkeeper, 2020-04-29:
Disable Adaptive Layers or reduce the Adaptive Layers Maximum Variation setting to get a layer thickness closer to the median Layer Height setting.

closed 2020-04-29 by Ghostkeeper