Ultimaker/Cura #7596: Stlicing does not start after pressing slice

muhammadelmogy, 2020-04-25:
**Application version**
4.6

**Platform**
Windows10 1909

**Printer**
Custom XYZ Cartesian

**Reproduction steps**
1.open model in cura
2.slice first time most likely will work
3.every other time press slice and wait, it will display processing
4.change view to "Prepare" or "Monitor" then button revert back to slice text
5.change the view again to "Review" click again on the slice button, slice will start and finish quickly

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
If I follow the steps till 3 and wait it will take long time to finish

**Expected results**
If steps followed it will behave as described otherwise  it will take too much time

**Project file**
[CFFFP_Fan Holder.zip](https://github.com/Ultimaker/Cura/files/4532162/CFFFP_Fan.Holder.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-04-28:
Please attach the logs, as per the instruction of the template. We really need it to debug things.

muhammadelmogy, 2020-05-01:
Log File Uploaded!

mahtDFR, 2020-05-01:
I'm afraid I can't reproduce it from your project file project file and your logs suggest it's slicing ok.
Are you able to try [resetting your configuration](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations)? You can zip up the relevant version number first to keep your settings safe.

`2020-04-28 14:25:21,312 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G28 ;Home
2020-04-28 14:25:21,314 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] 
2020-04-28 14:25:21,317 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] M190 S100.000000 ; Set bed temperature
2020-04-28 14:25:21,319 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] M109 S240.000000 ; Set nozzle temperature
2020-04-28 14:25:21,322 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G29
2020-04-28 14:25:21,324 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G1 Z1.0 ;Move the platform down
2020-04-28 14:25:21,327 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G92 E0
2020-04-28 14:25:21,329 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G1 Z1.0 ;Move the platform down
2020-04-28 14:25:21,332 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G1 F1000 E15
2020-04-28 14:25:21,335 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G1 F5000 Z10.0
2020-04-28 14:25:21,337 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G1 F3000 E-4
2020-04-28 14:25:21,340 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G1 F2000 E8
2020-04-28 14:25:21,343 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] G92 E0" -s acceleration_layer_0="3000" -s machine_max_feedrate_e="299792458000" -s travel_compensate_overlapping_walls_x_enabled="True" -s speed_wall_0="30.0" -s cool_lift_head="False" -s ooze_shield_angle="60" -s wireframe_flat_delay="0.1" -s smooth_spiralized_contours="True" -s machine_center_is_zero="False" -e0 -s cool_lift_head="False" -s retraction_prime_speed="70" -s machine_extruder_end_pos_abs="False" -s spaghetti_infill_extra_volume="0" -s support="0" -s retraction_min_travel="2" -s machine_nozzle_tip_outer_diameter="1" -s infill_material_flow="100" -s skin_overlap_mm="0.02" -s jerk_travel_layer_0="1" -s meshfix_maximum_deviation="0.025" -s infill_pattern="triangles" -s magic_fuzzy_skin_point_density="1.25" -s support_interface_skip_height="0.3" -s top_bottom_thickness="0.8" -s skin_angles="[ ]" -s support_tower_roof_angle="65" -s raft_surface_acceleration="1500" -s machine_nozzle_heat_up_speed="2.0" -s support_roof_line_distance="0.4" -s support_join_distance="2.0" -s machine_extruder_end_pos_x="0" -s jerk_support_infill="1" -s support_zag_skip_count="12" -s raft_surface_fan_speed="0" -s machine_heat_zone_length="16" -s prime_tower_wipe_enabled="True" -s support_use_towers="False" -s raft_fan_speed="0" -s extruder_nr="0" -s ironing_pattern="zigzag" -s acceleration_support_interface="1500" -s machine_steps_per_mm_x="50" -s speed_z_hop="10" -s slicing_tolerance="middle" -s roofing_material_flow="100" -s bridge_wall_material_flow="50" -s coasting_volume="0.064" -s retraction_hop_only_when_collides="False" -s acceleration_print_layer_0="500" -s wall_overhang_speed_factor="100" -s acceleration_roofing="1500" -s z_seam_position="back" -s support_interface_line_width="0.4" -s skirt_brim_line_width="0.4" -s bridge_skin_material_flow_3="110" -s machine_min_cool_heat_time_window="50.0" -s infill_angles="[ ]" -s wipe_repeat_count="5" -s optimize_wall_printing_order="True" -s material_break_temperature="50" -s support_tower_diameter="3.0" -s material_diameter="1.75" -s material_anti_ooze_retracted_position="-4" -s prime_tower_line_width="0.4" -s meshfix_maximum_travel_resolution="0.8" -s ironing_inset="0.2" -s layer_0_z_overlap="0.15" -s support_interface_density="100" -s sub_div_rad_add="0.4" -s machine_steps_per_mm_z="50" -s cross_infill_pocket_size="1.5" -s wipe_retraction_speed="40" -s machine_nozzle_offset_x="0" -s machine_nozzle_cool_down_speed="2.0" -s machine_settings="0" -s wipe_hop_speed="10" -s resolution="0" -s speed_equalize_flow_max="150" -s coasting_speed="90" -s small_hole_max_size="0" -s mesh_position_y="0" -s material_flow_dependent_temperature="False" -s acceleration_support_infill="1500" -s meshfix_union_all_remove_holes="False" -s brim_gap="0" -s support_bottom_line_width="0.4" -s support_bottom_pattern="concentric" -s support_interface_angles="[ ]" -s layer_start_x="0.0" -s support_fan_enable="False" -s top_thickness="0.8" -s raft_surface_line_width="0.4" -s wall_line_width_x="0.4" -s bridge_wall_speed="10" -s wipe_retraction_extra_prime_amount="0.1" -s ironing_only_highest_layer="True" -s meshfix="0" -s retraction_hop_after_extruder_switch="True" -s bottom_thickness="0.8" -s raft_interface_jerk="1" -s skin_line_width="0.4" -s command_line_settings="0" -s raft_base_acceleration="1500" -s support_brim_enable="True" -s switch_extruder_prime_speed="20" -s small_feature_speed_factor="50" -s support_interface_offset="0.0" -s blackmagic="0" -s acceleration_support_roof="1500" -s wall_material_flow="100" -s travel_compensate_overlapping_walls_enabled="True" -s wipe_retraction_prime_speed="40" -s switch_extruder_retraction_speed="20" -s jerk_ironing="1" -s r [... cut ...]
2020-04-28 15:05:56,181 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2020-04-28 15:05:56,183 - INFO - [MainThread] cura.CuraApplication.closeApplication [577]: Close application
2020-04-28 15:05:56,190 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2020-04-28 15:05:56,324 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [304]: Attempting to kill the engine process
`