Ultimaker/Cura #7597: Feature: ReadOnly Gcode Settings

JRRN, 2020-04-25:
As a Gcode User I want to can see in readonly mode configuration settings of a Gcode file.

The current status is that appears a message "Print Configuration is disables. You can't modify the GCode File".

[image]

fieldOfView, 2020-04-26:
Not all settings can be read from the gcode file. Some can be inferred. But there is not enough information in the gcode file to restore all settings.

JRRN, 2020-05-01:
If you open the gcode files, you can recover some settings about it:

;SETTING_3 {"extruder_quality": ["[general]\\nversion = 4\\nname = 3DN_Artillery
;SETTING_3 _Genius_Base\\ndefinition = artillery_genius\\n\\n[metadata]\\npositi
;SETTING_3 on = 0\\nintent_category = default\\ntype = quality_changes\\nsetting
;SETTING_3 _version = 11\\nquality_type = fast\\n\\n[values]\\nbrim_line_count =
;SETTING_3  10\\ncoasting_enable = True\\ncool_fan_full_layer = =max(1, int(math
;SETTING_3 .floor((cool_fan_full_at_height - resolveOrValue('layer_height_0')) /
;SETTING_3  resolveOrValue('layer_height')) + 2))\\ncool_fan_speed = =100.0 if c
;SETTING_3 ool_fan_enabled else 0.0\\ncool_min_layer_time = 10\\ndefault_materia
;SETTING_3 l_print_temperature = 205\\ninfill_enable_travel_optimization = True\
;SETTING_3 \ninfill_pattern = gyroid\\ninfill_sparse_density = 80\\nironing_enab
;SETTING_3 led = False\\nironing_only_highest_layer = True\\njerk_print = 7\\nje
;SETTING_3 rk_wall_0 = 6\\nmaterial_flow = 95\\nmaterial_flow_layer_0 = 110\\nma
;SETTING_3 terial_print_temperature = 195\\nraft_airgap = 0.1\\nraft_smoothing =
;SETTING_3  2\\nretract_at_layer_change = True\\nretraction_amount = 5\\nretract
;SETTING_3 ion_hop = 0.25\\nretraction_hop_enabled = True\\nretraction_speed = 4
;SETTING_3 0\\nroofing_layer_count = 1\\nskirt_line_count = 2\\nspeed_equalize_f
;SETTING_3 low_enabled = True\\nspeed_infill = 100\\nspeed_print = 20\\nspeed_wa
;SETTING_3 ll_0 = 35.0\\nspeed_wall_x = 35.0\\nsupport_angle = 40\\nsupport_infi
;SETTING_3 ll_rate = 3\\nsupport_pattern = triangles\\nsupport_tree_angle = 20\\
;SETTING_3 nsupport_xy_distance = =wall_line_width_0 * 2\\nsupport_z_distance = 
;SETTING_3 =layer_height if layer_height >= 0.16 else layer_height*2\\ntop_botto
;SETTING_3 m_pattern = zigzag\\ntop_bottom_thickness = 0.8\\nwall_line_count = 2
;SETTING_3 \\nwall_thickness = =line_width * 2\\nz_seam_corner = z_seam_corner_i
;SETTING_3 nner\\n\\n"], "global_quality": "[general]\\nversion = 4\\nname = 3DN
;SETTING_3 _Artillery_Genius_Base\\ndefinition = artillery_genius\\n\\n[metadata
;SETTING_3 ]\\nintent_category = default\\ntype = quality_changes\\nsetting_vers
;SETTING_3 ion = 11\\nquality_type = fast\\n\\n[values]\\nadhesion_type = skirt\
;SETTING_3 \ndefault_material_bed_temperature = 70\\njerk_enabled = True\\nlayer
;SETTING_3 _height = 0.3\\nlayer_height_0 = 0.3\\nmaterial_bed_temperature = 60\
;SETTING_3 \nretraction_combing = off\\nsupport_type = buildplate\\n\\n"}

Ghostkeeper, 2020-05-07:
Yes, that lists the changes that the user made (not in a default profile). It's being used by Ultimaker's support to see if the user changed anything weird. It should not list all 600 settings for every extruder, because that makes it unusable for a quick review, impossible to trace back which settings were changed by the user (and which came from a profile) and also breaks the functionality of importing the g-code as a custom profile.