Ultimaker/Cura #7599: 2D to stl

abtionlove, 2020-04-26:
What part of the code is used to convert PNG and other images into STL files in Cura? Can you take it out and use it alone?

fieldOfView, 2020-04-26:
See https://github.com/Ultimaker/Cura/tree/4.6/plugins/ImageReader, specifically https://github.com/Ultimaker/Cura/blob/4.6/plugins/ImageReader/ImageReader.py#L53

It would take quite a bit of work to make a standalone script out of it, especially considering the script does not write any files but it creates a Uranium SceneNode. But with some work you could make it produce a Trimesh node instead, and use the STL saving commands in Trimesh.

closed 2020-04-28 by nallath

abtionlove, 2020-04-29:
> 参见https://github.com/Ultimaker/Cura/tree/4.6/plugins/ImageReader，特别是https://github.com/Ultimaker/Cura/blob/4.6/plugins/ImageReader/ImageReader.py#L53
> 
> 用它制作一个独立的脚本将需要大量的工作，尤其是考虑到该脚本不会写入任何文件，而是会创建一个Uranium SceneNode。但是通过一些工作，您可以使其代替Trimesh节点，并在Trimesh中使用STL保存命令。

Thank you very much. It's not easy to implement. Is there any other way to transfer pictures to STL files? Thank you

Ghostkeeper, 2020-04-29:
You could load them in Cura and use File -> Export to save the STL if that's all you want.

Or you could use different software. There's plenty of software that does stuff like this, such as [Fusion 360](https://www.instructables.com/id/3D-CNC-Relief-Sculpture-Fusion-360/), [SketchUp](https://www.youtube.com/watch?v=s6EM1MQTlrw) or [specifically-designed software](https://sourceforge.net/projects/heightmap2stl/).

abtionlove, 2020-04-29:
> You could load them in Cura and use File -> Export to save the STL if that's all you want.
>
> Or you could use different software. There's plenty of software that does stuff like this, such as [Fusion [...]

Thank you very much. I want to implement this function on Android. Is there any good way?

Ghostkeeper, 2020-04-30:
I don't know of any. Feel free to take inspiration from our code if you're going to be building an app, as long as you comply to the license.