Ultimaker/Cura #7601: Infill line multiplier not display for Zig Zag pattern

acoder84, 2020-04-26:
**Application version**
4.6.0

**Platform**
Linux

**Printer**
Ender 3

**Reproduction steps**
1. Open any stl
2. Select Infill pattern Grid
3. Set **Infill line multiplier** value greater than 1
4. Change Infill pattern to Zig Zag

**Screenshot(s)**
[image]

**Actual results**
I cannot change **Infill line multiplier** because it has disappeared.

**Expected results**
I would expect Infill line multiplier will not hides for Zig Zag pattern.

**Project file**

**Log file**

**Additional information**

rburema, 2020-04-28:
We explicitly switched that off:
`"enabled": "infill_sparse_density > 0 and not spaghetti_infill_enabled and infill_pattern != 'zigzag'",`

smartavionics, 2020-04-28:
Although you can still see it when the pattern is lines and connect infill lines is enabled...

[image]

rburema, 2020-04-28:
That's an interesting pattern, although I would hesitate to call it 'zig-zag'.

Ghostkeeper, 2020-04-28:
This is because with the infill line multiplier it always converts the infill lines to polygons in order to prevent huge blobs of overlap. The contours are always fully connected in circles. This can't be combined with zig-zagging along the borders, since those borders are already connected by the multiplier.

Rather than disabling zigzag when the multiplier is used, we've opted to disable the multiplier when zigzag is used. But this is only front-end. In the back-end it then reverts to lines infill apparently. Maybe instead we should make it revert the infill line multiplier rather than the infill pattern?