Ultimaker/Cura #7602: When in preview mode and srcoll up and down with slider the screen turn blue . The colour of the support is the smae colour as the model to be printed .

Blockbusted, 2020-04-26:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-04-27:
Please fill in the issue template, or we won't (be able to) help.

Blockbusted, 2020-04-27:
Have done so .

nallath, 2020-04-28:
The template still isn't filled in. Could you try again?

Blockbusted, 2020-06-19:
The support material is the same color as the material that is being used
to print , henceforth one can't see clearly what is support or the model to
be printed ....

When you zoom in or out to see detail the model can not be seen as it
disappears at the bottom of the picture (screen).Hope this is more clear ..

jellespijker, 2020-06-19:
Can you please fill in the issue template

Blockbusted, 2020-06-20:
Hi 

I don’t know how to do that .If this is the only way to get help then we call it quitt’s.

 

Thank you for your time .

jellespijker, 2020-06-20:
[image]

jellespijker, 2020-06-20:
Just checking if you're familiar with the `color scheme` drop down in the preview mode

[image]

Blockbusted, 2020-06-21:
Thanking you for this information Jelle . Much appreciated .

jellespijker, 2020-06-21:
Did this solve your issue?

Blockbusted, 2020-06-21:
Hi
Yes this did resolve my query.
Thank you for your assistance .

closed 2020-06-21 by jellespijker