Ultimaker/Cura #7603: Creality Ender 3 Pro

Stuart1949, 2020-04-26:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Stuart1949, 2020-04-26:
ver 4.6 when Creality Ender 3 Pro only standard quality is supported.

octera, 2020-04-26:
the problem was solved here #7550

rburema, 2020-04-28:
Stay tuned for 4.6.1 (patch release) later this week.

closed 2020-04-28 by rburema