Ultimaker/Cura #7604: Update Profile with Current Settings buggy

ok4global, 2020-04-27:
**Application version**
Mac 4.6.0

**Platform**
Mac

**Printer**
UM3E

**Reproduction steps**
1. modified print setting
2. updated profile
3. ran slicer
4. modified more settings
5. updated profile, but new settings not saved, sometimes.

**Actual results**
New settings not saved.  Restart cura, then works.

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-04-28:
As I can't reproduce this, can you _please_ fill out the entire sheet and especially, attach the logs.

Ghostkeeper, 2020-04-28:
What do you mean, "new settings not saved, sometimes"?

ok4global, 2020-04-29:
Sorry @rburema , I don't know what logs you want.

I'll explain more thoroughly.

I use the printer to make prototypes of organic complicated sculptures — precision does not matter, but speed and aesthetics do.  To make one sculpture prototype, I need to print 3-5 parts.  Using the fastest default setting, it would take well over 24 hours to print, but I tweaked it so it only takes 15 hours.  Still, since it often finishes in the middle of the night, I usually only get to print one per day, so it takes 3 to 5 days to make on complete prototype.  Yuk.

So, I keep playing with the setting to try and speed it up.  Despite the page online showing the different infill options, there is no clear information about the the impact of infill pattern on speed, nor advice which infill can be the lowest percentage but still print the top correctly.  

So, I start with my custom profile, change the infill pattern and run the slicer, then change to the next one and run the slicer, then again and again until I see which infill saves me time.  The slicer takes a long time, so while it is slicing, I scroll up and down the settings list studying it without changing anything so I am ready with my next test as soon as the slicer is done.  Each time the infill is better than previous infills, I click on _Update profile with current settings_ so that I have saved the best one.  

I do the same thing with other settings like Conical Support, Gradual Infill, Fuzzy Wall, etc etc.  Each time running the slicer to see the impact, each time saving changes to current profile when they are good.  

If I have a saved profile which is working, I often want to check my changes against my last successful print because I can't remember if I my last profile had Gradual Infill Step Height set for 1.5mm or 2.5 mm, or if the angle of the conical support was 30 or 20.  So, since the only way to compare profiles is to switch between them, periodically I do so to look at previous settings.  

I habitually keep clicking on _Update profile with current settings_ like I used click save every 5 minutes in MS Word because it kept crashing, but sometimes I forget to save the settings before switching to a different profile.   

When that happens, Cura warns me and offers to let me _Keep or Discard changes_ or gives me the option to save a new profile — none of which I want.  I want to update the settings the current profile.  So, I click the little red X in the top right corner to close the dialog.  Then I click on _Update profile with current settings_.  Then I change profiles, but do not get the warning indicating that my most recent changes have been saved.  

[image]

By "new settings not saved, sometimes" I mean just that.  Yesterday I was playing with infill and prime tower.  My new profile was taking about the same time as my old profile, then suddenly it was much faster.  I noticed that the prime tower was gone, turned off, but my infill changes were saved.  There were other examples of some changes being saved, and others not.  It is not consistent.  

Also, in my experiments with slicing different infills, I noticed that the changes are not reflected in the re-sliced model (i.e. I change pattern in the settings but the when I slice it the pattern doesn't change).  Once that happens, I can't figure out any way to get it to accept the new pattern except restarting Cura.  Maybe it is all related.

thanks, Steve

nallath, 2020-04-29:
We understand what you're _trying_ to do. But we need the logs cura generates to see what is going _wong_. The template explicitly mentions this (it's actually still in your original post: https://github.com/Ultimaker/Cura#logging-issues)

ok4global, 2020-04-29:
Thanks for the link telling me where the logs are stored.

[cura.log](https://github.com/Ultimaker/Cura/files/4550579/cura.log)

Ghostkeeper, 2020-04-30:
> I habitually keep clicking on Update profile with current settings like I used click save every 5 minutes in MS Word because it kept crashing, but sometimes I forget to save the settings before switching to a different profile.

The changes that are not saved to a profile are saved to disk anyway, so even if Cura crashes it would still have saved them. Internally, the "unsaved" changes are a profile just as much as your custom profiles.

> So, I click the little red X in the top right corner to close the dialog.

This behaves the same as the "keep" button. You can tell it to always keep the changes and it won't pester you again about it. That's what I do as well.

In the log file I don't see any weird errors, or settings that get dropped or anything. It's quite a long log file though so maybe I'm missing something somewhere. The reproduction steps don't reproduce the issue for me though, so I'm guessing that maybe something weird is going on with different settings. My exact steps are these:

* Change infill density to 42%.
* Save changes to a custom profile.
* Load a model and slice it. It has 42% infill now.
* Change infill density to 1%.
* Update the custom profile with the current changes.
* Slice again. It correctly has 1% infill now.
* Change infill density to 90%.
* Update the custom profile with the current changes.
* Slice again. It correctly has 90% infill now.

Are you sure you're making the changes on the correct extruder?

The prime tower gets disabled automatically if there is only one extruder in the print, by the way. It won't need a prime tower if it doesn't need to switch extruders. Also for some materials the prime tower is disabled by default.

ok4global, 2020-04-30:
Thanks for trying to replicated it @Ghostkeeper.  When I do it on my machine, it is glitchy.  Don't know why.  Hopefully it is just me.  If no one else if having the problem, then no need to spend more time on it.  

Maybe it is because I was slicing and reslicing it a lot without saving each time in between because I wanted to see the impact of the change before saving it to the profile.  

FYI, my steps were:

- load model
- load custom profile
_- run slicers to see print time_
- change infill pattern
_- run slicers to see print time_
- change infill pattern again
_- run slicers to see print time_
- change infill pattern again
_- run slicers to see print time_
- change infill pattern again
_- run slicers to see print time_
- change infill density
_- run slicers to see print time_
- change infill density
_- run slicers to see print time_
- mess around with prime tower settings
_- run slicers to see print time_
- Update the custom profile with the current changes

I had two extruders on all the time, so that doesn't explain the prime tower turning off.  

Also, I use extruder 2 only for PVA interface layer between supports and model, so I never change those settings.  It is unlikely I was making changes to the wrong extruder.  

Thanks

Ghostkeeper, 2020-05-06:
After starting from scratch I added an UM3E printer, changed the infill density to 42% and saved that as a profile. Then I rebooted Cura to simulate your most likely steps (since you mention that this issue is affected by rebooting Cura). Then I executed your reproduction steps.

After the last step, "Update the custom profile with the current changes," I checked the settings that I had changed: Prime tower was still active, infill density still at 10% and infill pattern was still Cubic. All three were as I last left them. Also if I press slice again, it slices with those settings correctly, although it didn't produce a prime tower since my print was only using the first extruder in the print.

So I'm still not able to find the bug on my side :(

closed 2020-06-19 by no-response[bot]

scottjames, 2020-12-19:
if it helps, I was also not able to save to the original default profile.  But if duplicate and save my own profile, I am able to update that one.  I think this helps to protect the original factory/stock profile, but allow modification of users' profiles.