Ultimaker/Cura #7606: Prints settings for dual extruder / make dual column

ok4global, 2020-04-27:
**Is your feature request related to a problem? Please describe.**
The way the current print settings is laid out is very complicated.  Some features affect all print heads, some affect only the selected print head.  It is not intuitive.

Also, it is very hard to compare the unique settings between print heads. it requires switching back and forth between tabs and trying to remember the settings.  Yuk.

**Describe the solution you'd like**
Instead of tabs for each extruder, it would be much easier if there were two columns.  

Any setting which is shared between extruders can be a single cell which spans both columns.  It would be a simple and intuitive indication that the setting is shared.

Any setting which is unique to each extruder would have two cells, one in each column.  It would be a simple and intuitive and would allow for a quick comparison between the settings for each extruder.  

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
everyone

**Additional context**
(Add any other context or screenshots about the feature request here.)

rburema, 2020-04-28:
You can already see which is which: The little chain-link icon just to the left of the settings fields (where the f(x) icon and the back-arrow icon are also sometimes) indicates that the setting is shared between extruders.

Ghostkeeper, 2020-04-28:
Duplicate of https://github.com/Ultimaker/Cura/issues/7359.

closed 2020-04-28 by Ghostkeeper

ok4global, 2020-04-28:
> You can already see which is which: The little chain-link icon just to the left of the settings fields (where the f(x) icon and the back-arrow icon are also sometimes) indicates that the setting is shared between extruders.

Yes, exactly: the small icon exists but It is not user friendly, simple nor intuitive. A small modification to the GUI would make it intuitive and also easy to compare the settings between extenders without having to flip back and forth between different screens.