Ultimaker/Cura #7608: Doesnt load

DeividasLT, 2020-04-27:
Cura 4.6 after instaling new update in first it work and now it doesnt start
windows 10

**Printer**
(Creality Ender 3 pro)

nallath, 2020-04-28:
Could you pretty, pretty please fill in the bug template? We simply can't help you if you don't fill in the _entire_ bug template.

DeividasLT, 2020-04-28:
**Application version**
4.6

**Platform**
Windows 10 Pro 64-bit
Intel Hd graphics 2gb
**Printer**
ender 3 pro

**Reproduction steps**
1.opened cura

**Screenshot(s)**
[image]

**Actual results**
Cure.exe is not responding error

Ghostkeeper, 2020-04-28:
We still can't really help. The template also contains an entry about your log file. This is very relevant for your issue.

You have to understand that for us so far it's been starting up fine. We believe you on your word, but it's not like we just "forgot to make it load" or something so we need a bit more information to know why your case is different from ours and hopefully enough information to be able to cause it also not to load for us, so that we can track down what makes it not load and then fix it.

DeividasLT, 2020-04-29:
Sorry im just really new in to this type of stuff so sorry for inconvenience
[cura.log](https://github.com/Ultimaker/Cura/files/4554374/cura.log)

Ghostkeeper, 2020-05-04:
Hmm, okay I have two possible explanations:
* In your last 2 runs (the ones that seem to be failing to start) I don't see CuraEngine starting at all. This is started asynchronously so it kind of runs through the start-up procedure at the same time. Is it possible that a firewall is blocking CuraEngine.exe and then freezing Cura.exe since it's making a local socket connection?
* You might also give it some time. In the last run it seems to be busy removing hidden container stacks, which is started by one of the built-in packages in its init stage (hence the message on screen). This was necessary because of a bug in Cura 4.4 that caused the printer to be duplicated in the background every time you switched printers within a printer group. If you have a lot of those, it might take a few seconds. Not really more than a minute though, I'd say.

DeividasLT, 2020-05-04:
okey, just now i tried to turn of firewall same problem and i waited for aprox 5 min it still doesn't turn on and even tried with 4.6.1 update still same problem, looking via task manager she even doesn't consume any memory even when starting and loading

Ghostkeeper, 2020-05-07:
Hmm, well, thanks for trying and providing the log, but I'm sort of out of ideas what could cause this then! I haven't seen this bug myself yet. Don't forget to turn the firewall back on ;)

There is one more thing then that might shine light on the issue. Rather radical. Could you go to `C:\Users\<your username>\AppData\Local\cura` and put the whole `4.6` folder in a zip archive, and post that here?

This zip archive contains your entire state of Cura pretty much. It includes all of your installed plug-ins, your profiles, the printers you added and the settings that are changed in each of them, your log and your preferences.

If that doesn't reproduce it, the only difference left is the computer hardware and OS. We'd be looking at a very difficult and weird bug if that were the case.

closed 2020-06-19 by no-response[bot]