Ultimaker/Cura #7609: Bottom View

scotriddle, 2020-04-27:
There's no actual bottom view.  And rotating the view to examine the bottom of a build is squirrely, awkward and hard to control to the desired perspective.

Ghostkeeper, 2020-04-28:
There is no back view button either. We left those two out on purpose because it's not so common there is anything interesting to see on those sides. Overloading the interface with buttons is also a concern of usability.

Please fill in the issue template that you so graciously erased. It contains an entry about why the issue is important, why we'd need to reconsider our earlier choice.

mahtDFR, 2020-04-29:
I've made an internal ticket for this here after some discussion around it: CURA-7410

@scotriddle As above, it sure would be useful if you could use our template, otherwise we have to come back and ask you to do it again.

fieldOfView, 2020-04-29:
A way to do this without adding new buttons is to change the top view button to a bottom view button when the view is the top view. So from any other view, the top view button would behave the same it does not, but after clicking it it becomes a toggle between top and bottom view (until the user rotates the camera away). The same could be done with the front & back views.

scotriddle, 2020-04-29:
I'm obviously new to this forum stuff.  I rarely have anything worthy of posting.  I've pasted the template for "New Feature" as suggested by the Keeper of Ghosts. I don't know if this is what you wanted me to do but here goes:

**Is your feature request related to a problem? Please describe.**
I'm using Cura 4.6.0.  It's getting the job done for me.  I just wish I could see the bottom straight on, like the Top View button does.  I'm more than happy to rotate the view around to the bottom myself but when I get to that upside-down orientation - or even from a top down orientation, actually - the Rotate View (RMB+drag) function doesn't rotate like it does immediately following clicking on the "3D VIEW" button.  It seems to favor 'gimballing' on the polar axis.  Also, the "Zoom toward mouse direction" doesn't work, regardless of the combination of features selected under "Preferences>Viewport behavior" or whether the model is selected or not in "Prepare" mode. (Sorry, should that be in a separate ticket?)

**Describe the solution you'd like**
I've been using Solidworks for 14 years so I'm used to its view rotation characteristics.  Maybe I'm spoiled on Dassault but if Cura could manipulate the view more like it does, that'd be great.  

**Describe alternatives you've considered**
A "Bottom" button would be cool in lieu of tweaking the rotating functionality.  Maybe a "Preference>View" setting for different CAD software...?

**Affected users and/or printers**
Examining the bottom of my sliced model in Cura makes it really easy to see Support material problems that will add build time and possibly mess up the build.  I don't use bed adhesion with my CR-10, i.e., Brim, Skirt, etc.  I coat the glass with a product available at Home Depot and it works great with PLA and TPE.  I haven't found ABS is necessary for what I'm modeling but it doesn't lift off at recommended nozzle and bed temp settings using this product.

**Additional context**
I like fieldOfView's suggestion but I have plenty of room on my monitor for a few more buttons, if you choose to add any.

scotriddle, 2020-04-29:
Great idea!  Thanks, Aldo.

scotriddle, 2020-04-29:
I did the template thing.  I hope I did it right.  Thanks, maht.

mahtDFR, 2020-04-29:
> I did the template thing. I hope I did it right. Thanks, maht.
 
Much appreciated, thanks.

> (Sorry, should that be in a separate ticket?)

If possible make each issue a separate ticket, yes. That way it can be tracked by us.

scotriddle, 2020-04-29:
No prob on both.  Should I open a "Type: Bug" or "Type: Improvement" on the Zoom issue?

mahtDFR, 2020-04-29:
> Should I open a "Type: Bug" or "Type: Improvement" on the Zoom issue?
If it's not behaving as expected it's a bug.

scotriddle, 2020-04-29:
K, goddit. Thanks!

Ghostkeeper, 2020-04-30:
We've received reports about the zoom to mouse functionality sometimes not working. I've seen it myself as well, but it only happens sometimes and it's still a mystery to us.

We've discussed adding a bottom view button with the team and came up with a use case of needing to inspect the support more easily, which is a valid reason to have one. We're not sure yet how the icon should look, but we'll figure that out. It's on our backlog now (CURA-7410 for developers).

scotriddle, 2020-05-01:
Thank you all for your time and consideration.  For me, a Bottom View button would be no more than a shortcut to quickly get to that underneath perspective (of course).  It isn't a necessity or a deal breaker for me.  I'm sure there are more pressing and important issues you guys are dealing with.

Keep u the great work!

eddieparker, 2021-04-19:
I also would love this for support reasons.  Is there any reason to not at least add them to the View -> Camera Position drop-down?  It would be great to have at least Bottom in there, although since it's a drop down I don't see why you don't just have back in there as well.

Ghostkeeper, 2021-05-06:
Yeah, don't see harm in just adding it to the menu for now. Not in the main window, just the menu.

closed 2021-05-06 by Ghostkeeper