Ultimaker/Cura #7611: unique Benchy defect with Cura 4.6 beta or 4.6 final

bill-orange, 2020-04-27:
Application version: 4.6 beta and 4.6 final
Platform: Windows 10
Printer: Tronxy X3A
Reproduction steps: 

1. Print Benchy with gcode from any earlier version. Version 4.1 is a good choice.
2. Print Benchy with gcode from 4.6
3. compare top view

Screenshot: included.  
Actual results: Observe the void in the top edge of the motor compartment and smokestack
Expected results: no voids:
Project file: Included
Log file: not relevant

Additional information: I have printed an armada of Benchys  in attempts to rid myself of stringing.  My latest Benchys made iwth Cura version 4.6 show type of defect that I have not seen before.  Unfortunately, I have NOT done comprehensive regression testing to eliminate any other possible factors.

Is this a setting issue or a genuine BUG?

[image]

[stag_7_65_35_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/4540182/stag_7_65_35_3DBenchy.zip)

bill-orange, 2020-04-27:
Here's an additional photo for this bug report.  The image on the left is from Cura 3.1.  The Image on the right is from 4.6.
[image]

mahtDFR, 2020-04-28:
Possibly related to #7565 
We're going to make a patch release this week

Ghostkeeper, 2020-04-28:
> Possibly related to #7557
> We're going to make a patch release this week

It's not. You get the same result in 4.6.1.

Ghostkeeper, 2020-04-28:
I manually performed an inverse of the version upgrade from 4.5 to 4.6 on your project file and loaded it in 4.5, but it shows the same effect.

[image]

Maybe you have a project file from 4.5 (or any earlier version as you say) that produces the desired result?

As measured in Blender, the thickness of this wall is exactly 2mm. After the inset for the outer wall that becomes 1.6mm (the width of the polygon traced by the centreline for the outer wall). After the inset for the first inner wall that becomes 0.8mm. After the inset for the second inner wall that becomes 0mm, thus there is no more second inner wall.

You can resolve the issue by reducing the inner wall line width to 0.397mm (3μm lower). Normally you'd be able to reduce it by just 1μm and you'd bet a 2μm-wide polygon which effectively gives one single line and a travel move back, with wall overlap compensation. However to reduce the effect of rounding errors with properly tessellated thin walls in curves, we have a [heuristic](https://github.com/Ultimaker/CuraEngine/blob/69f0ca961b44a258c4e9e9ebf628ada044c75433/src/WallsComputation.cpp#L72-L96) that tries slightly different line widths to see if the number of parts is reduced, so 3μm it is then. At that sort of scale, rounding to micrometres makes a real difference.

Ghostkeeper, 2020-04-28:
The experimental variable line width feature would resolve this sort of issues, hopefully: https://github.com/Ultimaker/CuraEngine/pull/1210 It's still a long way off though.

bill-orange, 2020-04-28:
The problem was not present in Cura 3.x  I never noticed this problem on Cura 4.4.

Asterchades, 2020-04-29:
Try enabling "Fill Gaps Between Walls". I'll admit I never really tried any Cura before 4.0 (I'm only a recent inductee to the 3D Printer owners' society) but I do know that setting was required to seal the box when I tried my own Benchies.

bill-orange, 2020-04-30:
@Asterchades  You are correct. Setting "Fill Gaps Between Walls" to "Everywhere" corrects, or at least papers over) the problem.  I don't know if @Ghostkeeper  wants to consider this the solution or consider it a temporary fix until a more permanent solution is implemented.

Ghostkeeper, 2020-05-06:
> (why is this a drop-down and not a check box?)

Historical :smile: We should change it one day.

Ghostkeeper, 2020-05-06:
As far as I'm concerned the current system will pretty much always have this problem. I wouldn't want a fix for this specific case. If you're willing to spend some time on tweaking settings, I strongly advise you to [try a few line widths](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/resolution/line_width.md#adjusting-line-widths-to-fit-enough-walls) to get a better fit. Otherwise just use the gap filling technique.

The gap filling technique is a general solution to this problem. It's intended precisely for this sort of issue. We know that the gap filling technique has some issues though, especially with thin curved walls where it sometimes produces a lot of small, separate line segments, so we're reworking that to use a wholly different technique.

bill-orange, 2020-05-06:
Thanks. Good advise.

closed 2020-05-06 by bill-orange