Ultimaker/Cura #7612: Unknown setting causing 10mm at y max and min to be reserved

clfaye, 2020-04-27:
**Application version**
4.6

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
Cura is reserving 10mm at y min and y max for no reason that I can determine.  Build Plate Adhesion is set to none - so it's not the reservation normal for a skirt.

**Screenshot(s)**
[image]

mahtDFR, 2020-04-28:
This is because the machine profile has specified this to allow space for the clips on the bed. See discussion here: https://github.com/Ultimaker/Cura/issues/7252

clfaye, 2020-04-28:
This was indeed a no go area in the definition file.  If you delete that limit the bars go away.

closed 2020-04-28 by clfaye