Ultimaker/Cura #7613: Blurry text Cura 4.5, and 4.6

KelpyGsus, 2020-04-27:
**Application version**
(The version of the application this issue occurs with.)
Cura 4.5 beta, 4.5, 4.6 beta, 4.6.

**Platform** 
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Windows 10 Version 10.0.018362 Build 18362
GTX 1080, driver version 445.75

**Printer**
(Which printer was selected in Cura?)

Ender 3

**Reproduction steps**

Not really repeatable, just constantly blurry. 
Blurry text regardless of light mode, or dark mode. 
I uninstalled and reinstalled multiple times, tried version 4.5 beta, 4.5, 4.6 beta, and 4.6.
Text is normal in the preferences tab.

**Screenshot(s)**

https://imgur.com/a/6GGZtuy
Top screenshot is Cura 4.4. The rest are 4.6

**Actual results**

Nothing changes.

**Expected results**

N/A

**Project file**

N/A

**Log file**

[cura.log](https://github.com/Ultimaker/Cura/files/4542326/cura.log)

**Additional information**

N/A

mahtDFR, 2020-04-28:
Possibly related to this? https://github.com/Ultimaker/Cura/issues/5586

fieldOfView, 2020-04-28:
What strikes me as odd is that the titlebar text and menu item text in your screenshots looks suspect too:
[image]

The top 2 are yours, the bottom one is mine.

Ghostkeeper, 2020-04-28:
Yeah that's really weird. Are title bars for other applications blurry as well?

KelpyGsus, 2020-04-29:
> 
> 
> Yeah that's really weird. Are title bars for other applications blurry as well?

No, Cura 4.5 and above is the only time I've had this issue. 4.4 and below work fine.

KelpyGsus, 2020-05-03:
It was an issue with Windows ClearType text.

closed 2020-05-03 by KelpyGsus