Ultimaker/Cura #7614: [4.6.0] Non-Ultimaker Printers not listed

mnswamp1, 2020-04-28:
**Application version**
4.6.0

**Platform**
Windows 10 OS version 1903 build 18362.778

**Printer**
None - that's the issue in a nutshell

**Reproduction steps**
1. New install of Cura 4.6 from Ultimaker website.  (New Cura user, too)
2. Completed install, accepting defaults.
3. Launch Cura.
4. Preferences - Printers - Add printer
 
** Screenshot**
[image]
6. Tried to twist toggle, etc.  
7. uninstall/reinstall - no joy

** Expected result**
select Creality Ender 3 printer

**Log files** 

[stderr.log](https://github.com/Ultimaker/Cura/files/4543065/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/4543066/stdout.log)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-04-28:
There is a scrollbar, but it is hidden untill you use it.

Ghostkeeper, 2020-04-28:
Duplicate of https://github.com/Ultimaker/Cura/issues/6299.

closed 2020-04-28 by Ghostkeeper

mnswamp1, 2020-04-28:
Wow.  You really, really have to dig for that one.  Thanks so much for quickly ID'ing the issue.