Ultimaker/Cura #7615: Preview layer by layer not working, Cura 4.6

kornshadow097, 2020-04-28:
If you enable horizontal expansion for holes and traditional (ex. o.64mm H.E. / -0.35mm Hole H.E)
Then the Preview mode wont work. It just stays dark grey and layers are no longer available for view 

**Application version**
(The version of the application this issue occurs with.)
4.6
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10
**Printer**
(Which printer was selected in Cura?)
tevo tarantula
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
just adjusted H.E on holes and x/y
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

kornshadow097, 2020-04-28:
https://mega.nz/file/HnAxlYhJ#pZsaTvrcoZB1kp4F6UwUW-S_BvUQSwFkYT3UyqZfB4Q

key is at the end of the link

rburema, 2020-04-28:
The (non active) layers staying grey is a known issue. (Your print settings is currently in front of it but) If you move the layer-selector slider in the far mid-right, then the colours should come back.

As for the support blocking bug, can you send us a project file (Save, don't Export, rename to appease github when attaching).

rburema, 2020-04-28:
(P.S. Screenshots can be embedded on github reports ... just drag them to your writeup when writing it.)

kornshadow097, 2020-04-28:
I moved the layer selector to the middle to show it didn't affect the view.
No matter where the slider is placed, it looks identical. The only place
colors appear are the very top layer.

Yea I'll save the file whe I get a chance.

kornshadow097, 2020-04-28:
Good to know. Thanks

rburema, 2020-04-28:
If you're only seeing the top layer, you're not dragging the layer-selector. I mean the vertical slider on the right, not the horizontal one on the bottom.

closed 2020-06-19 by no-response[bot]