Ultimaker/Cura #7617: No sale el desplegable de configuración de la temperatura

pakocc18, 2020-04-28:
Hola.

Quiero configurar las temperaturas de trabajo, pero no se despliega la configuración de MATERIAL, en el programa cura 4.6.0

¿Es un error de la versión? en la versión anterior lo hacía sin problema.

Saludos

rburema, 2020-04-28:
I'm sorry, but I can't reproduce this. I just see this when I click on 'Manage Materials...'

[image]

Can you tell me what the steps you took?

pakocc18, 2020-04-28:
El error no es ahí, eso puedo acceder sin problema.
El error es en el ajuste de la impresión.

No se despliega el menú de ajuste. Lo marco en rojo
[](url)
[image]

Ghostkeeper, 2020-04-28:
Could you click on this little gear, next to the Material header:

[image]

And make sure that the settings you want to see are visible?

Also, which printer are you using?

It would help to communicate in English. I learned Spanish out of interest for 2 months on Duolingo but never got really good at it.

pakocc18, 2020-04-29:
Hello.
I will try to explain myself in English.

I do have some settings activated. Attached images
My 3d printer is ultimaker 2 extended +

Thanks for your attention and interest

[image]
[image]

pakocc18, 2020-04-30:
Tranks