Ultimaker/Cura #7618: [4.6.0] Layer Skipped

Doriangaensslen, 2020-04-28:
**Application version**
4.6.0

**Platform**
[image]

**Printer**
Ender 3

**Reproduction steps**
1. Download https://www.thingiverse.com/thing:790533/zip
2. Select only one CE3_Top_CornerF2 File
3. Print Settings Layer height 0.2 Infill Density 55% Speed 60mm/s
4. Print

**Screenshot(s)**
[image]
[image]

**Actual results**
The printer Prints flawlesly but just "forgets" to print a layer. I've tried again and it stopped ad the exatly same height (Picture 1). When i tried to print the same piece again but with other parts, it printed without a problem. Today i printed another piece with complete other settings from thingiverse, and it skipped again a layer on another height (Print 2). Normally i would suspect the printer but i've double checked everything, z-travels light and without a problem, the extrusion isn't and wasn't a problem. Since its reproducable with a file and the same settings i belive its a problem with the Slicing.

**Expected results**
Print without problems

**Project file**
See description

**Additional information**
-none-

bill-orange, 2020-04-28:
If you look in Preview does the preview show the layer missing or with printing accomplished correctly?

Doriangaensslen, 2020-04-28:
In the preview the layers are perfectly fine

bill-orange, 2020-04-28:
Thanks. I thought it might be the same problem as #7611

Ghostkeeper, 2020-04-29:
> 4. Print

This is where I get stuck with the reproduction steps, since I don't have a Creality Ender 3.

From what I can tell from your pictures, comparing them to my own attempt at following your steps, it seems to fail around layer 90, no? I don't really see anything weird around layer 90. The closest thing that could possibly be a mistake is that it resets the E value at layer 119 because it would exceed E5000 and then cause larger and larger rounding errors in some firmware. But maybe the `G92 E0` occurs somewhere else for you. It's hard to tell without a project file or the g-code if the slice is exactly the same for me.

Doriangaensslen, 2020-05-01:
I will print this piece again and if it fails again il send you the Gcode. right now im waiting for filament, since ive used all.

Ghostkeeper, 2020-05-06:
Or maybe the Z stage gets stuck? Could use a bit of lubricant?

GregValiant, 2023-09-15:
Is this still applicable today?  Can this be closed now?

closed 2023-09-30 by github-actions[bot]