Ultimaker/Cura #7619: Preview sliders no longer working as expected

wnypoker, 2020-04-28:
4.6.0
(The version of the application this issue occurs with.)

OSX - 10.14.6 (18G4032)
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Ender 3
(Which printer was selected in Cura?)

**Reproduction steps**
1. With any file, slice object
2. Switch to preview, horizontal and vertical sliders no longer work.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

Attempts to drag Vertical and Horizontal sliders fail.  Occasionally clicking in the text field of the vertical slider allows you to change the value.
(What happens after the above steps have been followed.)

Sliders should allow you to move thru the various layers of the object.
(What should happen after the above steps have been followed.)

Issue appears with any project file.
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

Can't find any reference to issue in the log file.
[cura.log](https://github.com/Ultimaker/Cura/files/4548382/cura.log)
[image]

(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

KenLucke, 2020-04-28:
Data point:  This works fine for me, functions as always.  Cura v4.6.0, Mac OS 10.15, MacBook Pro.

Ghostkeeper, 2020-04-29:
Hmm, strange. Also works fine for me.

Do you have any plug-ins that could affect this, such as the SideBar GUI plug-in? If so, could you try disabling those and trying again?

KenLucke, 2020-04-29:
> Hmm, strange. Also works fine for me.
> 
> Do you have any plug-ins that could affect this, such as the SideBar GUI plug-in? If so, could you try disabling those and trying again?

Another data point :  I am using the Sidebar Plugin, and it still works fine.