Ultimaker/Cura #7627: Cura 4.6.1

Metalbenji, 2020-04-29:
**Application version**
4.6.1

**Platform**
WIndows 10, GTX1070, ryzen processor

**Printer**
ender 3

**Reproduction steps**
1. opened program
2. left idle

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
pc freezes have to pull power and hard reset

**Expected results**
should not crash, 4.6 had no issue at all could leave it open for hours 4.5 would crash all the time too

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-04-30:
Please fill in the entire template. That includes adding logs.

Ghostkeeper, 2020-05-04:
Duplicate of https://github.com/Ultimaker/Cura/issues/5265.

closed 2020-05-04 by Ghostkeeper