Ultimaker/Cura #7628: [4.6.1] Crash when rtying to delete faulty Profile

FlailAway, 2020-04-29:
**Application version**
4.6.1, 4.6.0

**Platform**
Mint 19.3 +Cinnamon

**Printer**
None involved

**Reproduction steps**
1. Tried to use a Profile that was fine yesterday.
2. When I click on it, it loads "Draft"
3. Went to "Manage Profiles" and click on the errant Profile
4. Cura tosses a big message box Copied as much as I could below.
5. Close that message and all of Cura slams shut
6. Cannot send Report as PC does not have Internet access and I am using a VNC to operate it.

**Actual results**
See #5 above

**Expected results**
(What should happen)
I should have been given an option by Cura to auto-delete the errant Profile before slamming shut, if it must.

ALSO, there should be a method to manually navigate to the Profile folder and delete the errant one manually. I looked in .local/Share/Cura/4.6 and could not make any sense out of the file names so had no idea which f the 30-profiles to delete. 

Dare I suggest, you need a stand-alone bulk Profile manager.

ALSO, the the top "box" of the error message needs to be a Read-Only TMemo so I can also copy and paste the information here to help you.

Here is as much of the error messages I can copy
*********************************************************************Traceback (most recent call last):
  File "/build/4.6/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualitySettingsModel.py", line 67, in setSelectedQualityItem
  File "/build/4.6/build/inst/lib/python3.5/site-packages/cura/Machines/Models/QualitySettingsModel.py", line 104, in _update
KeyError: 'id'

Thread 0x00007f4ab385e700 (most recent call first):
  File "/tmp/.mount_Cura-4wjqKNS/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ad0ff9700 (most recent call first):
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 161 in _storeStderrToLogThread
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ad17fa700 (most recent call first):
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 149 in _storeOutputToLogThread
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ad27fc700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ad2ffd700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ad37fe700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ad3fff700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007f4b051b7740 (most recent call first):
  File "/build/4.6/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 379 in _logInfoWidget
  File "/build/4.6/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 180 in _createDialog
  File "/build/4.6/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/build/4.6/build/inst/bin/cura", line 163 in exceptHook
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/build/4.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 813 in run
  File "/build/4.6/build/inst/bin/cura", line 231 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.6/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.6/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

nallath, 2020-04-30:
Oh! You actually have that error we saw in the crashes but could never reproduce / figure out how it happend. Great! 

Could you pretty please send us your zipped configuration folder? You might have the key that we need to _finally_ fix this issue.

FlailAway, 2020-05-01:
> Could you pretty please send us your zipped configuration folder? You might have the key that we need to _finally_ fix this issue.

I have good news and bad news. 
The bad news is it is working OK again this morning.
The good news is I know what made it good, but certainly not why or how.

This morning I clicked a known good Profile (A) and it dropped "Draft" in it's place. Brief moment of Deja-Vu or panic so I clicked another Profile (B) that was similar and it locked in OK. I went to Manage Profiles to rename (B) by adding "pla," to the front of the name before changing anything. 

I closed Manage Profiles and for some weird reason thought I'd check that (A) again and dang me, all good and printing with it right now!

Are you sure the config folder will be of real use to you? It is on a neighbor's PC and I cannot get the File Access via the VNC and neighbor not savvy enough to change Root-level attributes for me from afar. 

I really do not want to go visit. ;)

FlailAway, 2020-05-01:
> Could you pretty please send us your zipped configuration folder? You might have the key that we need to _finally_ fix this issue.

What files do you need and what are the Paths to them? I may go to their place and make a quick grab if you can be specific.

nallath, 2020-05-04:
We need the folder that you see when you select help -> show configuration folder. 

I'm not sure if we will be able to fix the issue if we have the config folder, but we're probably not able to fix it without. Reproducing the issue is the first step in being able to fix it after all.

Ghostkeeper, 2020-05-04:
This is the crash that gets reported via that crash dialogue the most, aside from one crash in the Thingiverse plug-in which is not really under Cura's control.

Somehow there is a profile without container ID in Cura's memory. That should never happen and as far as I could find all possible code paths to constructing a container cause it to have an ID. But it definitely occurs in the field.

On Linux, the folder we need is `~/.local/share/cura/4.6`. Probably only the files related to the current printer and profiles you were using would be enough, but to be certain it would be better to save everything. It should only be a couple of kilobytes anyway unless you installed a lot of plug-ins (and if file size is a problem you might also just tell us which plug-ins were installed).

FlailAway, 2020-05-04:
OK, it is still doing it but with a different Profile each day. I can get the `~/.local/share/cura/4.6` for you. But, with no idea what actual data is in it, I do no want to upload it to the general public as this is not MY data and I will not risk my neighbor's data on an open forum.

Can you not set up an Upload folder here on GitHub for the Public to upload to, but cannot be Viewed and/or Downloaded by the Public? I can stretch to trusting you guys, but the Public et al, not so much.

Ghostkeeper, 2020-05-06:
You can send it to r dot dulek at ultimaker dot com if you like.

FlailAway, 2020-05-11:
Sorry for the delay, been a bit busy. Only 2-plug-ins, I'll try to get over there in a day or so and grab the 4.6 Folder

Marketplace - Installed
Custom Supports 0.0.4
Setting Visibility Set Creator 3.5.2
No Material has been installed.