Ultimaker/Cura #7629: sliced file has inverted X and y axis direction

zorro80306, 2020-04-29:
I am not sure if this is a bug.  I am running Cura 4.5 with a Fusun 3D printer.
I like to know if there are setting configuration or sw modification that I can make to invert the X/Y axis direction when slicing the model. The problem I have is the model display correctly in the 3D and sliced/preview display, but the model printed in x and y axis direction. when I tried to move the print head via the manual control under the monitor view, all axis moving properly in all directions according to the "arrows" buttons.

The funny thing here is the 3d model displayed correctly in both the file loaded and preview windows.  All manual controls from the "monitor" windows also working properly.  , When printing the model, directly via USB interface (or generated G code file), the part is in flapped image along the X and Y axis.

Ghostkeeper, 2020-05-04:
> I like to know if there are setting configuration or sw modification that I can make to invert the X/Y axis direction when slicing the model.

There is not. Not really. Cura always assumes a left-handed coordinate system as it's shown in the interface.

If the X and Y axis are swapped, you can "fix" that by mirroring the X coordinate and then rotating the model 90 degrees around the Z axis.
If the X and Y axes are both mirrored, you can "fix" that by rotating your model 180 degrees.

I've never heard of a printer with a right-handed coordinate system though. That sort of system is common with computer graphics but not really anything else as far as I'm aware.

Are you sure it's not just a lithophane that you're holding upside down or something?

Ghostkeeper, 2020-05-04:
Ah, nevermind. I see that this is a duplicate of https://github.com/Ultimaker/Cura/issues/7630 and you fixed the problem there in your firmware.

closed 2020-05-04 by Ghostkeeper

zorro80306, 2020-05-04:
> 
> Ah, nevermind. I see that this is a duplicate of #7630 and you fixed the problem there in your firmware.

Hi, The fix I posted, by changing the FW, is just a way I get around the problem.  The bottom line here is the Cura 4.5 SW seems has problem corresponding on the arrow control and home buttons operating direction in opposition to what the 3d displayed on the build envelop windows.   Yes, mirroring the model before sliding can also over come the same issue, but I believe it will make the SW mus more user friendly and WYSWYG if it gives the user an option to invest the X/Y axis direction before slicing.  I hope you will agree with me.  Best regards