Ultimaker/Cura #7630: Invert X and y axis of the sliced output

zorro80306, 2020-04-29:
I am not sure if this is a bug.  I am running Cura 4.5 with a Fusun 3D printer.
I like to know if there are setting configuration or sw modification that I can make to invert the X/Y axis direction when slicing the model. The problem I have is the model display correctly in the 3D and sliced/preview display, but the model printed in x and y axis direction. when I tried to move the print head via the manual control under the monitor view, all axis moving properly in all directions according to the "arrows" buttons.

The funny thing here is the 3d model displayed correctly in both the file loaded and preview windows.  All manual controls from the "monitor" windows also working properly.  , When printing the model, directly via USB interface (or generated G code file), the part is in flapped image along the X and Y axis.

konskarm, 2020-04-30:
This doesn't really sound like a problem on Cura's side, since we haven't had reports about that before in any other printer. Could it be that something is not calibrated with your printer? Also, I cannot find a Fusun 3D printer reference.

ozligia, 2020-04-30:
@zorro80306 Do you have origin point the same place as Cura does? (lower left corner).
If you have point of origin in some other courner, it can be that printer will move "according to arrows", but the axis will be inverted.

zorro80306, 2020-04-30:
Hi, Konsharm,
Thanks for replying to my question so quickly.  Yes, you are correct, I don’t think this is a Cura problem either.  

After I did more experiments, here is my conclusion.  It looked like the X and y axis steppers on my printer is set up in a different direction than the slicer expected.  This also applies to how my X/y end stop is set up in the Marlin FW.  Yes, I properly can overcome this issue by reconfiguring the Marlin FW and relocating the end stop SW positions.  I do, however, think it could be much easier to fix the problem if there is a way to modify Cura and instruct it to invert the X/y axis while slicing/generating the G code.  By the way, I did not experience the same issue on the Z axis.

With the current setup of my printer, I can manually, from the Cura/Monitor window, control the movement of both X/Y axis in the direction corresponding to the “arrow” keys layout and both “home” buttons are working properly.  The only thing is the 3D display of the model is in its mirrored (both X/Y) image of the actual part.  If I set Cura to mirror the part (both on X/Y axis) first then slice and print the part will print out correctly.  The problem I described here also applies to the 3D display (both at the “prepare” and “Preview” windows. 

I hope the information I provided her helps to explain a bit more about what I am trying to accomplish.  Once again thanks for your expert advise .

Best regards

zorro80306, 2020-04-30:
> 
> 
> @zorro80306 Do you have origin point the same place as Cura does? (lower left corner).
> If you have point of origin in some other corner, it can be that printer will move "according to arrows", but the axis will be inverted.

@ozligia
The origin of my Cura is located at the lower left corner when the SW started anth the X/Y arrows movement controls are working in corresponding to the arrows direction.  One thing I have to point out here is, I did have to reconfigure my printer's Marlin firmware to invert the drive direction on both the X/y axis to make it work as I just described.  Thanks!

[image]
.

zorro80306, 2020-04-30:
@ozligia
By the way, if what you described how the Cura origin point and "Arrow" control direction works is true, I believe this could be a bug of Cura.  The "Arrow" control should always work corresponding the display orientations.  Otherwise it will be very confuse to the Cura user on what we are actually working on (WYSWYG).  Agree?

konskarm, 2020-05-01:
The arrows (e.g. when moving the object) do indeed correspond to the display orientations (WYSWYG). You can see that when moving the object using the arrows, in the coordinates of the object after moving it.
[image]

@zorro80306 I am glad that your issue is fixed through reconfiguring the Marling FW. The use case for an "Invert X/Y axis" setting sounds very small and of limited use. Considering that Cura already has 550+ settings, we want to avoid adding anything that wouldn't be frequently used.

mahtDFR, 2020-05-01:
I don't think this is something that should be fixed in Cura, as it's up to the user to configure their firmware. So I will close this.

closed 2020-05-01 by mahtDFR

zorro80306, 2020-05-01:
> 
> 
> I don't think this is something that should be fixed in Cura, as it's up to the user to configure their firmware. So I will close this.

Hi, thanks for the feedback, but I do think this is indeed a bug.  Please refer to @ozligia's comment.  
Yes, rewiring my print is one way to over come this problem.  By the way, it will take just reconfigure the Marlin FW.  I actually also need to relocate the X/YU end stop switch position and modify the Marlin FW.  It is a kind of complicated process.  I do think it is a much better solution if there is a way we can add a setting in Cura to allow "easy fix".  This added future can make Cura easier and more user friendly as a "Open Source" platform form any printer builder.  I hope you will agree with my opinion here.   
Best regards,

Ghostkeeper, 2020-05-06:
There's also a fix you can do in your start g-code: Simply set the steps per mm to something negative with `M92`. For instance, if 200 steps equate to 1mm of movement in your printer, add this to your start g-code:

    M92 X-200 Y-200

No need to move any hardware around then. This is precisely the sort of printer-specific thing that the start g-code is meant for.

zorro80306, 2020-05-06:
> There's also a fix you can do in your start g-code: Simply set the steps per mm to something negative with `M92`. For instance, if 200 steps equate to 1mm of movement in your printer, add this to [...]

Hi, Thanks for the additional instruction.  Yes, I can use you suggestion to make the printing work.  I do. however, still think it will be a nice new feature Cura should consider and make the SW more user friendly and for more printer platform.  Here are my other reasoning:
A. It is kind of hard for the user to detect the "flopped" problem, because the the actual 3D display of          the model is in reverse orientation to what is actually printing out.  The way how the "arrow" movement controls work and the 3D display of the model is not really in complainant of the concept of WYSWYG. 
B. I am not sure if there is a setting in Cura that allows me to add the start g-code.  I looked over all the printer sittings in Cura 4.61 and not finding any.  If this is true, the only way I can add the start code is first generate the g-code file and manually add the cod in the the file and then reload the file to print.  This also means I can no longer just load up the model, slice and start printing directly from Cura. 

Once again, I hope the info. I provided here is helpful to make Cura a better SW for more user and printer support.  I sincerely  hope you and the developer member of Cura will re-consider and "add" the X/Y axis inversion function as one of the standard setups in Cura.
Thanks again for taking your time to respond to my comments.  I look forward to continue helping to make Cura the better SW.  Best regards.

zorro80306, 2020-05-06:
@Ghostkeeper
Oh, By the way, I just found where to add the "startup g-code".  Thanks for the tips.

KleaTech, 2020-05-06:
Please consider reopening and implementing this feature. Normally the origin of the printer is at front-left. But on my printer it's at rear-left. Please provide a way to specify the origin or flip x/y axis.

Ghostkeeper, 2020-05-06:
I don't think we will. This is precisely what the start g-code field is for.

Yes, it's possible to edit the start g-code in the interface in the screen where you edit the printer itself, but we really consider this to be part of the printer's inherent properties (like the printer's width, height and depth in the same dialogue), and we expect those printer settings to be written by people who design those printers, or at least are adept at it. In that field you can write down whatever your firmware needs configured to get started. If your printer has a different coordinate system than what Marlin has or other conventional firmwares, then you can configure that there too.

There are hundreds of different things you can configure in Marlin and thousands of things that printers do differently if they use different firmware. We're not going to make switches for them all. That's why we have an arbitrary field of start g-code.

Jeffwill611, 2021-10-04:
I just recently got a Creality Ender 5 Pro. It prints great, but the default home position is the back right corner. Cura seems to assume that the home position position is the front left. I can find no way to configure this. The effect is that the model prints facing the back instead of the front as shown in Cura. I'm fairly new to 3D printing, so I'm really not interested at the moment in having to redo any firmware. 

I've read through this thread and wonder if Cura is just hard wired to the front left on assumption. Seems Creality has changed the home position for their printer. I've seen videos and read pages to indicate that this might be the case with all Ender 5 Pro printers. Unless they are were mislabeled from the factory, I've confirmed that the X and Y servos are wired correctly. 

Is this specific to just the Ender 5 Pro? I ask because I did notice that Cura doesn't have a listing for this printer, but does natively support Ender 5 and Ender 5 Plus. For those that have experienced this issue, do they have better results with other slicers?

Much appreciated!