Ultimaker/Cura #7632: UI enhancement - prepare - indicate bed adhesion present

tablatronix, 2020-04-30:
## Overview

Add a visible indication on the prepare view that there is a `bed adhesion` option set thus affecting the max size of the print area.

### Problem
Almost 100% of the time I have something that doesnt fit and is greyed out, it is because I have a bed adhesion aka skirt or something on, and can never fix it until i remember ( since you cannot slice and see it! ), I can only imagine how much bad UX this causes new users!

### Options
Automatically calculate max print boundary with and without bed adhesion calculations, and if bed adhesion causes out of bounds, add some alert indicating this is causing the overflow

If there is a bed adhesion set, simply indicate it somehow make something red, borders of bed, gray areas.. Something.

Actually render the item in prepare also

mahtDFR, 2020-04-30:
Seems like a good idea, this has caught me a few times too. Perhaps just a pop up will do for now, colour scheme changes might get complicated. Let's see what the others think.

mahtDFR, 2020-05-04:
There are actually quite a few settings that would influence the allowed area besides adhesion, e.g. horizontal expansion. We have an existing ticket in our system already to solve this. But I can't find it.

Ghostkeeper, 2020-05-04:
Maybe we should add a dark border around the build volume to indicate that the build volume has been reduced in size by the build plate adhesion :smile: 

The full list of settings that influences the grey zones is this: