Ultimaker/Cura #7633: Ironing only on top view surfaces

tomaswcz, 2020-04-30:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

There is a problem if you enable the ironing not only for the highest layer. The slicer irons also internal layer of the print. Try with https://www.thingiverse.com/thing:277394

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

I believe that the perfect solution for this would be to iron only layers that are visible from TOP perspective

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Alternative solution would be to ask user to select concrete surfaces that need to be ironed, the problem is that this would require some interactive function where the user would have to select the surfaces. Not easy to implement.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

Everyone would benefit from this, the ironed surfaces look very nice.

**Additional context**
(Add any other context or screenshots about the feature request here.)

mahtDFR, 2020-04-30:
If I understand correctly the indented layers are being ironed as well? It would be useful if you can send a project file with this, then I can check with your exact settings.

One workaround for the issue if I've understood it correctly is to use a cutting mesh overlayed on the top layers. You can then enable ironing for these layers only.

fieldOfView, 2020-05-01:
Won't that workaround introduce extra walls?

tomaswcz, 2020-05-01:
The workaround is not clear to me, but here you can see what I am talking about. If I don't leave the "Iron Only Highest Layer" option on, there are some places whre the layer is being ironed inside?

Highest layer only checked:
[image]

Unchecked:
[image]

Ghostkeeper, 2020-05-04:
What you are seeing there was a bug in 4.6.0 that got fixed in 4.6.1: https://github.com/Ultimaker/Cura/issues/7572

The original feature request still stands though.

However I think this just an example of a case where you specifically want to indicate "I want it there, there and there, but not there and there". The case of "everything visible from a perfect top-down view" is very model specific. There will be many models that want to iron on various types of surfaces, so this is not really a solution. Just something you'd build for one specific print.

It's meant to be a list of tunings. Not a maze through which the user has to find a combination that works for him. I don't think we should build this.

tomaswcz, 2020-05-04:
Hello, yes, maybe you are right that the implementation is not worth it.

If its a bug as you have described, then it is still there in 4.6.1, because that is the version I used for the screenshots.

Ghostkeeper, 2020-05-06:
Oh, that doesn't happen for me then:

[image]

Perhaps a project file would help?

tomaswcz, 2020-05-06:
Well thats strange. Maybe it another problem. Since I "got the update" I have two version of cura installed (4.5 and 4.6.1). The problem is there in my 4.6.1, but maybe it uses dependecies of 4.5?

The project file

[AA8_Ripple_test.zip](https://github.com/Ultimaker/Cura/files/4586141/AA8_Ripple_test.zip)

Ghostkeeper, 2020-05-06:
Ah, I see it happening too now, with that project file. The problem was not occurring for me because I was using the ZigZag pattern, which mistakenly uses the normal skin line width to see how far from the edge it should stop and turn around, rather than the line width or spacing of ironing. This is in progress right now: https://github.com/Ultimaker/CuraEngine/pull/1251

Concentric then reveals a bug (soon to be revealed with ZigZag too) where somehow it thinks there is a difference between two adjacent layers while there really isn't.