Ultimaker/Cura #7634: trying to add a non-network printer in 4.6.1, only Ultimaker BV printers listed

VOCAPTAIN, 2020-04-30:
following guidance from your ultimaker support page on how to add a custom printer to Ultimaker Cura

**Application version**
(The version of the application this issue occurs with.)
4.6.1 and 4.6.0
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows 10 pro 64, Intel i32310M
**Printer**
(Which printer was selected in Cura?)
n/a
**Reproduction steps**

Following guidance from your ultimaker support page on how to add a custom printer to Ultimaker Cura

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
[image]

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-04-30:
You can scroll the list of printers. There's a scrollbar that is hidden until you use it.

fieldOfView, 2020-04-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/7614

VOCAPTAIN, 2020-04-30:
thanks silly me the hidden scroll bar got me :-)

closed 2020-04-30 by VOCAPTAIN