Ultimaker/Cura #7636: 3rd party printer definitions on Linux cannot find "quality" settings in home directory

BenBergman, 2020-04-30:
**Application version**
4.5.0

**Platform**
Arch Linux 5.6.7, AMD RX480 GPU

**Printer**
Artillery Sidewinder X1

**Reproduction steps**
1. Install 3D Nexus printer definition for Artillery Sidewinder X1 in `~/.local/share/cura/4.5/hresources` as described [here](https://3d-nexus.com/newsroom/news-announcements-m/3dn-artillery-sidewinder-x1-and-genius-cura-4-5-machine-and-profiles-installation-package-released)
2. Open up Cura and add a non-networked Artillery Sidewinder X1 printer
3. Attempt to import the `.curaprofile` files from the above link.

**Screenshot(s)**
This is the expected outcome after importing profiles (a workaround was used). Prior to the workaround, the profiles section was empty (no defaults, no custom profiles)
[image]

**Actual results**
Importing the profile complained about missing default profiles.

**Expected results**
Importing profiles should result in new custom profiles available for use.

**Project file**
N/A

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4559566/cura.log)

**Additional information**
By copying the `resources/quality` files into `/usr/share/cura/resources/quality` I was able to work around the issue, but it would be nice to be able to keep the custom configs out of the way of the offifcially installed files.Okay,

Ghostkeeper, 2020-05-06:
These profiles are copied from the Creality profiles. As such they have the same filenames as those Creality profiles. Cura uses the filename as a unique identifier. As such it confuses these profiles with the Creality profiles. It finds that the profile `base_global_standard` already exists (as it loaded that from `/usr/share/cura/resources/quality/creality/base/base_global_standard.inst.cfg`) and then doesn't load the second one. Then it finds that there is no standard-quality global profile for the Artillery Sidewinder.

This is not a bug, but a problem with those profiles. Normally Cura prevents creating duplicate names when you create profiles, but when you mod them into the internal storage directory for Cura's configuration there is nothing we can really do to stop you from messing up the profile database.

Arguably the Creality profiles should not have been called "base_global_standard" and such since those names are not specific to the printers they are specific to. But that is a different issue.

closed 2020-05-06 by Ghostkeeper

BenBergman, 2020-05-06:
Ah, good to know. Thanks for the info!

Ghostkeeper, 2020-05-11:
Renaming the files in that archive will probably work fine though, if they have names that don't collide with any other profile file names.