Ultimaker/Cura #7637: Print Sequence - On at a time

MiguelFigueiredo47, 2020-04-30:
Hello,
I am Miguel Figueiredo
I just update Cura for 4.6.1 version.
I can NOT print sequence in One at a time. Cura reduce my build volume. its impossible to print. 
This issue was not happening before at old versions. 
How can i solve it?

**Application version**
Cura 4.6.1

**Platform**
Microsoft Surface

**Printer**
Ender 3

Vandrasc, 2020-05-01:
Hi @MiguelFigueiredo47 , thank you for your feedback. 

Unfortunately, I am not able to reproduce with just the information you provided, can you add a project file, screenshots, and logs? Thank you1

MiguelFigueiredo47, 2020-05-01:
> Hi @MiguelFigueiredo47 , thank you for your feedback.
> Unfortunately, I am not able to reproduce with just the information you provided, can you add a project file, screenshots, and logs? Thank you1

Hello @Vandrasc,
I am used to print sequecence one at a time. 
with this new version 4.6.1 is not possible.
Cura enable print sequence one at a time just with one object at the build plate. if I multiply my objects cura automaticly reduce my build volume.
- Print Sequence - One at a time.
[image]

-Print Sequence - One at a time (Multiple Objects) (PROBLEM)

[image]

mahtDFR, 2020-05-01:
Looks like the build volume height has been reduced. So it's not a problem with the XY plane, but the Z plane. Note the bounding box z limit around your objects - it cuts off at around 30mm.

Here - I reproduced it too. 

[image]

As a workaround you can increase the gantry height in the machine settings. By default it's set to 25 for some reason.

[image]

I've seen this before and I think it's a problem with the machine definition itself. This is a third-party definition so it's nothing to do with us. We try to avoid changing third party definitions because we can't test them, so the original contributor or somebody else with one of these machines will need to fix it and submit a **fully tested** version to us - then we can pull it into a future release.

Ghostkeeper, 2020-05-06:
The gantry height for the Ender 3 has been 25mm since forever. We have no reason to believe that to be incorrect. I don't understand why Miguel says this is a regression in 4.6.

MiguelFigueiredo47, 2020-05-07:
I would like to say was my mistake confirm this problema was a regression in 4.6. 
serious apologies for my mistake. I had print sequence one at the time before, but i have to admit was with lower sice 25mm height.
one more time sorry for create this subject.

closed 2020-05-07 by mahtDFR